modelica/ModelicaStandardLibrary #2286: PartialTwoPhaseMedium FluidConstants should be replaceable

thorade, 2017-08-01:
PartialTwoPhaseMedium redeclares FluidConstants, but not as replaceable. 
Media extending that interface can thus not redeclare FluidConstants anymore.
https://github.com/modelica/Modelica/blob/maint/3.2.2/Modelica/Media/package.mo#L5788-L5791

Could the lines be changed to the following (note the `replaceable`):
```modelica
partial package PartialTwoPhaseMedium
    "Base class for two phase medium of one substance"
    extends PartialPureSubstance(redeclare replaceable record FluidConstants =
Modelica.Media.Interfaces.Types.TwoPhase.FluidConstants);
```

The constant I would like to add is the (specific) gas constant, because not all equations are fitted to the "true" value as available from Modelica.Constants.R
If making the record replaceable is undesired, then I will just store that constant elsewhere.

Note: Dymola 2018 does not complain when doing a pedantic code check, but OpenModelica warns me that a record is being redeclared that is not replaceable.

beutlich, 2017-08-06:
@thorade You already can provide a PR for such few-line-changes, such that the library officers can review the changes and react faster.

beutlich, 2018-01-21:
Resolved in master.

closed 2018-01-21 by beutlich