modelica/ModelicaStandardLibrary #2291: createDirectory fails with relative directory

modelica-trac-importer, 2017-08-08:
**Reported by @claytex on2017-08-07T08:48:07+01:00**

Modelica.Utilities.Files.createDirectory does not work with relative directory for example:

```
Modelica.Utilities.Files.createDirectory("")
Not possible to create new directory
"C:/Users/Public/Documents/Dymola":
File exists
Failed to expand ModelicaInternal_mkdir(ModelicaStrings_substring("C:/Users/Public/Documents/Dymola", 1, 32)).
Failed to expand Modelica.Utilities.Files.createDirectory("").
```
----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2196

modelica-trac-importer, 2017-08-08:
Here is a suggested solution;
Modify createDirectory as follows:

```mo
algorithm 
  // Ignore call, if directory exists
  fullName := Modelica.Utilities.Files.fullPathName(directoryName);
  if not existDirectory(fullName) then
```

MartinOtter, 2017-08-08:
Function createDirectory(..) is called with an empty directory name. To my understanding, this should always be an error. The error message could be more clear, but the reaction is fine.

closed 2017-08-08 by MartinOtter