modelica/ModelicaStandardLibrary #2292: Race condition in tmpnam, called by ModelicaInternal_temporaryFileName

beutlich, 2017-08-08:
ModelicaInternal_temporaryFileName calls tmpnam which has known security issues, see e.g., https://stackoverflow.com/questions/5586355/c-getting-a-temporary-file-cross-platform and the links therein.

I cannot think of any fix, other than documenting the race condition in Modelica.Utitilities.Files.temporaryFileName.

MartinOtter, 2017-08-08:
It is just stated "tmpname is unsafe" or "dangerious", and that probably no portable other solution exists. It does not make sense to cite such sentences in the MSL documentation ("someone on StackOverflow claimed it is unsafe"). If not more information is given why this ticket is opened, I propose to close it.

beutlich, 2017-08-08:
I propose to a similar warning as stated at https://www.gnu.org/software/libc/manual/html_node/Temporary-Files.html: Warning: Between the time the pathname is constructed and the file is created another process might have created a file with the same name using tmpnam, leading to a possible security hole.

Furthermore, it should be added that the temporary file is not automatically removed once it is closed, but needs to be removed manually.

MartinOtter, 2017-08-08:
Feel free to add something of this kind. However, whenever someone works with any type of files in a simulation system, all kinds of problems can occur, whether temporary files or specific named files are used. Why should this be discussed in a supporting library for the modeling system?

beutlich, 2017-08-08:
> Feel free to add something of this kind.

Will do.

> However, whenever someone works with any type of files in a simulation system, all kinds of problems can occur.

Sure. Just check the closed issues on Modelica.Utilities.Files.*.

> Why should this be discussed in a supporting library for the modeling system?

To make the underlying C implementation as transparent as possible for the user. Typically, the user (if at all) reads the Modelica documentation, but does not look for the underlying C implementation.

sjoelund, 2017-08-08:
It's not just the website that says it's dangerous. We also get warnings from GCC whenever we link ModelicaExternalC or FMIL because of these issues with tmpnam. The function can cause seriously bad security bugs and there is not really any way to use the function in a secure way. That said it's hard to fix it in MSL because files are manipulated without using something similar to file objects...

⁣Sent from BlueMail ​

beutlich, 2017-08-08:
Resolved by abb8ec399e04bfa867990bc92b966b1c149d11b7 in master.

closed 2017-08-08 by beutlich