modelica/ModelicaStandardLibrary #2295: calloc(0, x) not well defined

DagBruck, 2017-08-11:
There is C code that uses the construct `calloc(0, x)` - i.e. to allocate zero elements of some size. However, you must allocate at least 1 element to be guaranteed a de-referenceable and free-able pointer. The effect of allocating 0 elements is not terribly well defined in C.

I found this is ModelicaMatIO.c, it may occur elsewhere too.

I suggest all such cases be changed to allocate at least one (dummy) element.

beutlich, 2017-08-11:
This is an upstream issue of [matio](https://github.com/tbeu/matio/blob/master/src/mat5.c#L3318) which Cppcheck reports as bug every time I run it: `Invalid calloc() argument nr 1. The value is 0 but the valid values are '1:'.`. It will be fixed once a new version of matio is released and and merged into ModelicaMatIO.

FYI: See #2001 for a another upstream issue of ModelicaMatIO of MSL v3.2.2.

beutlich, 2017-10-22:
Resolved in master by 164e1c2cfe5ad2170c0d08862041279e12f60fe6.

closed 2017-10-22 by beutlich