modelica/ModelicaStandardLibrary #2301: Use defaultComponentName to shorten instance names in Diagram layer

christiankral, 2017-08-18:
In #2215 it was discussed that the default component names of some models shall be shortened in order to fit the length constraints of text field containing ``%name`` in the Icon layer. This is particularly useful for components with very long class names.

christiankral, 2017-08-18:
I will be volunteering to change the default component names of the following packages:
- Modelica.Electrical.Analog
- Modelica.Electrical.Machines
- Modelica.Electrical.MultiPhase
- Modelica.Electrical.PowerConverters
- Modelica.Electrical.QuasiStationary
- Modelica.Magnetic.FluxTubes
- Modelica.Magnetic.FundamentalWave
- Modelica.Magnetic.QuasiStatic.FundamentalWave

christiankral, 2017-12-31:
Default component names are -- by now -- shortened for: 

-    Modelica.Electrical.Analog
-    Modelica.Electrical.Machines
-    Modelica.Electrical.MultiPhase
-    Modelica.Electrical.PowerConverters
-    Modelica.Electrical.QuasiStationary
-    Modelica.Magnetic.FluxTubes
-    Modelica.Magnetic.FundamentalWave
-    Modelica.Magnetic.QuasiStatic.FundamentalWave

I will also apply shorter default component names for

- Modelica.Mechanics.Rotational
- Modelica.Mechanics.Translational

as I am a library officer of these sub-packages, too.

beutlich, 2018-01-02:
There is no need to apply these enhancements to the maint/3.2.2 branch.

christiankral, 2018-01-02:
In addition to shortened component names, I tried to unify the default naming of sensors and sources. The following examples show the underlying "rule":

Sensor examples

- ```iSensor``` for a current (i) sensor
- ```vSensor``` for a voltage (v) sensor
- ```sSensor``` for a position (s) sensor
- ```tauSensor``` for a torque (tau) sensor

Source examples

- ```iSource``` for a current (i) source
- ```vSource``` for a voltage (v) source
- ```sSource``` for a position (s) source
- ```tauSource``` for a torque (tau) source

HansOlsson, 2018-01-02:
> •iSensor for a current (i) sensor
> •vSensor for a voltage (v) sensor
> •sSensor for a position (s) sensor
> •tauSensor for a torque (tau) sensor

I am not too fond of that naming. The names i,v,s,tau are too short on their own, and relies on the implementation details, and graphically I find that currentSensor, voltageSensor are more descriptive.

I understand that we do not want too long names and controlledIdealIntermediateSwitch was a mess - but to me this is going too far in the other direction.

Obviously this depends on the level of expertise with the library. I would expect someone unfamiliar with the library to be more helped by the slightly more descriptive names, and I think we should make it more friendly for the beginners when using the library.

I also realized another issue related to this - will open separate ticket.

christiankral, 2018-01-02:
I agree, that more descriptive names are better to understand. I came up with this naming "rule" when I stumbled into something like "magneticPotentialDifferenceSensor" which is as awkward as "magneticVoltageSensor". I cannot omit the term "magnetic" here, since voltage is usually associated with electric voltage. Do have a proposal for such awkward namings?

HansOlsson, 2018-01-02:
> I agree, that more descriptive names are better to understand. I came up with this naming "rule" when I stumbled into something like "magneticPotentialDifferenceSensor" which is as awkward as [...]

I don't know if there is a common term in the domain, and thus I only see two non-awkward solution: the long name - and just "sensor". In both cases users can override the default by renaming the component - and hopefully find a name based on the specific usage.

christiankral, 2018-01-02:
>I don't know if there is a common term in the domain, and thus I only see two non-awkward solution: the long name - and just "sensor". In both cases users can override the default by renaming the component - and hopefully find a name based on the specific usage.

OK, I now implemented ```magVoltageSensor``` as a short form of the magnetic voltage sensor. I think it is a compromise between descriptive name and the abbreviation "mag" to shorten "magnetic". Consequently, I removed all other default component names of sensors and sources, so most of them appear as they have been before implementing my proposal.

christiankral, 2018-01-02:
By going through the different source and sensor models of the different Modelica domains I noticed, that they are implemented all differently. For the next major release we could try to unify the naming of sources, sensors and parameters. In order to do so, we should possibly start a discussion on unified naming of sources, sensors, parameters (and other components and variables). The discussion will then require the development of guidelines being more strict than the ones listed in ```Modelica.UsersGuide.Conventions.ModelicaCode.Naming```.

christiankral, 2018-01-02:
So I am done with the default component names of the listed Modelica packages. If some other library officers are willing to provide shorter default component names for their packages, they could use this ticket to document the progress of their work. 

I will thus keep this ticket open.

tobolar, 2018-01-31:
I can add my experience based on working with the MultiBody library for years. To my opinion, the default component name should 
1. be as short as possible,
2. best reflect the component purpose and
3. respect the indexing rules as known from books.

Point 2. particularly means that the user should change as less as neccessary when instantiating the component. Therefore, "voltageSensor" is better then "sensor", esp. to avoid annoying renaming of the instances where e.g. sensors from different domains are used within a class. Using some superordinate property ("sensor") could yet be reasonable and helpful.
Point 3. says that "sensorVoltage" is better then "voltageSensor". This is helpful e.g. when "grouping" similar instances within the class for better understandability, or for post-processing, or when searching for an instance.
Finally, 1. says that "body" is still better then "partBody" and "spring" better than "forceSpring".

I'm quite sure someone can better formulate those rules/ideas but this is my contribution to the discussion.