modelica/ModelicaStandardLibrary #2302: Use consistent units in Modelica.Electrical.Spice3

christiankral, 2017-08-18:
In Modelica.Electrical.Spice3 some units are not declared correctly -- even if the units are obvious. 

First example (more may follow):

The units of the following variables are not specified correctly in ``Modelica.Electrical.Spice3.Internal.Functions.resDepTemp ``
- tc1 -> 1/K
- tc2 -> 1/K2
- difference -> K
- dCond_dTemp -> 1/K/Ohm2

tbeu, 2017-08-20:
Duplicate of #2087 which is duplicate of #414.

beutlich, 2019-03-20:
@christophclauss See again https://github.com/modelica/ModelicaStandardLibrary/issues/414#issuecomment-272599220 by @casella. Do you think adding proper units is feasible for MSL v4.0.0?

christophclauss, 2019-03-22:
Dear Thomas,

it is still an extensive part of work, but it should be feasible for MSL
v4.0.0.

Best regards

Christoph

christophclauss, 2019-08-04:
The last comment was wrong. I do no more believe in solving this issue before MSL4.0.0.