modelica/ModelicaStandardLibrary #2304: MyComplex vs. Complex

beutlich, 2017-08-21:
Left over from #1122: 
> But why not use Complex directly like the other packages already do?

Utilize the record Complex (and defined operators) instead of defining own record Modelica.Media.Air.ReferenceMoistAir.Utilities.Ice09_Utilities.Basic.MyComplex and functions MyComplexF.* to avoid redundancy.

Note, that there is Modelica.Media.Air.ReferenceMoistAir.Utilities.Ice09_Utilities.Basic.MyComplexF.'log' but no equivalent function in operator record Complex.

svanman1970, 2017-09-30:
hacktoberfest

AHaumer, 2018-05-18:
Why do you direct this question to Martin and me?
Why don't ask the library officers of Modelica.Media?

casella, 2018-05-29:
Conceptually, this is a trivial fix, but in practice it is quite tedious and it is easy to mess up. Furthermore, I don't see any test model in the MSL to check if the results are the same.

@wischhusen would you be available to take care of this issue (possibly also for the next minor release)?

beutlich, 2019-05-13:
> @wischhusen would you be available to take care of this issue (possibly also for the next minor release)?

@wischhusen Please let me know if you can manage for MSL 4.0.0?

beutlich, 2019-10-03:
> @wischhusen Please let me know if you can manage for MSL 4.0.0?

I prepared #3131 for it.

PR #3131 (2019-10-21): Replace Modelica.Media.Air.ReferenceMoistAir.Utilities.Ice09_Utilities.Basic.{MyComplex, MyComplexF} by operator record Complex
Closes #2304.
closed 2019-10-21 by beutlich