modelica/ModelicaStandardLibrary #2305: Invalid hyperlinks in Modelica help documentation

d94pn, 2017-08-21:
Scanning the help directory (help) for invalid links reveals several:

Modelica_Blocks_Examples_NoiseExamples.html: Invalid <a href>: modelica://Modelica.Blocks.Noise.GenericNoise
Modelica_Blocks_Examples_NoiseExamples.html: Invalid <a href>: modelica://Modelica.Blocks.Statistics
Modelica_Blocks_Examples_NoiseExamples.html: Invalid <a href>: modelica://Modelica.Blocks.Statistics
Modelica_Blocks_Math.html: Invalid <a href>: modelica://Modelica.Blocks.Examples.NoiseExamples.WeibullNoiseProperties
Modelica_Blocks_Math.html: Invalid <a href>: modelica://Modelica.Blocks.Examples.NoiseExamples.WeibullNoiseProperties
Modelica_Blocks_Math.html: Invalid <a href>: modelica://Modelica.Blocks.Statistics.Variance
Modelica_Blocks_Math.html: Invalid <a href>: modelica://Modelica.Blocks.Examples.NoiseExamples.WeibullNoiseProperties
Modelica_Electrical_Machines_Thermal.html: Invalid <a href>: modelica://Modelica.Electrical.Machines.BasicMachines.DCMachines.DC_Compound
Modelica_Electrical_Machines_UsersGuide.html: Invalid <a href>: modelica://Modelica.Electrical.Machines.Utilities.SwitchedRheostat
Modelica_Electrical_Machines_UsersGuide.html: Invalid <a href>: modelica://Modelica.Electrical.Machines.Utilities.RampedRheostat
Modelica_Electrical_Machines_Utilities.html: Invalid <a href>: modelica://Modelica.Electrical.Machines.Utilities.FToDQ
Modelica_Electrical_PowerConverters_Interfaces_Enable.html: Invalid <a href>: modelica://Modelica.Electrical.PowerConverters.Interfaces.Enable1
Modelica_Electrical_PowerConverters_Interfaces_Enable.html: Invalid <a href>: modelica://Modelica.Electrical.PowerConverters.Interfaces.Enable2
Modelica_Electrical_PowerConverters_Interfaces_Enable.html: Invalid <a href>: modelica://Modelica.Electrical.PowerConverters.Interfaces.Enable1m
Modelica_Electrical_PowerConverters_Interfaces_Enable.html: Invalid <a href>: modelica://Modelica.Electrical.PowerConverters.Interfaces.Enable2m
Modelica_Electrical_QuasiStationary_MultiPhase_Basic.html: Invalid <a href>: modelica://Modelica.Electrical.QuasiStationary.SinglePhase.Basic.Inductance
Modelica_Magnetic_FluxTubes_Material_HysteresisEverettParameter.html: Invalid <a href>: modelica://Modelica.Magnetic.FluxTubes.Shapes.GenericHystPreisachEverett
Modelica_Magnetic_FluxTubes_UsersGuide_Hysteresis.html: Invalid <a href>: modelica://Modelica.Magnetic.FluxTubes.Examples.Hysteresis.TransformerWithHysteresis
Modelica_Magnetic_QuasiStatic_FundamentalWave_BasicMachines_Components.html: Invalid <a href>: modelica://Modelica.Magnetic.QuasiStatic.FundamentalWave.BasicMachines.Components.QuasiStionaryAnalogWinding
Modelica_Math_Nonlinear.html: Invalid <a href>: ftp://ftp.inf.ethz.ch/doc/tech-reports/1998/306.ps
Modelica_Math_Random_Generators.html: Invalid <a href>: modelica://Modelica.Blocks.Noise.GlobalSeed.random
Modelica_Mechanics_Rotational_Components.html: Invalid <a href>: https://www.fmi-standard.org
Modelica_Mechanics_Rotational_Components.html: Invalid <a href>: https://www.fmi-standard.org
Modelica_Mechanics_Rotational_Examples.html: Invalid <a href>: https://www.fmi-standard.org
Modelica_UsersGuide_Conventions_Documentation_Format.html: Invalid <a href>: <li> The hyperlink text in between <code>&lt;a href=&quot;...&quot;&gt;</code> and <code>&lt;/a&gt;</code> should include the full main package name.</li>
Modelica_UsersGuide_Conventions_Documentation_Format.html: Invalid <a href>: &lt;a href="modelica://Modelica.Mechanics.MultiBody.UsersGuide.Tutorial.LoopStructures.PlanarLoops"&gt;
Modelica_UsersGuide_Conventions_Documentation_Format.html: Invalid <a href>:   &lt;a href="modelica://Modelica.Electrical.Machines.BasicMachines.AsynchronousInductionMachines.AIM_SquirrelCage"&gt;
Modelica_UsersGuide_Conventions_Documentation_Format.html: Invalid <a href>:   can be found in &lt;a href="modelica://Modelica.UsersGuide.Conventions.UsersGuide.References"&gt;[Gao2008]&lt;/a.&gt;
Modelica_UsersGuide_Conventions_UsersGuide.html: Invalid <a href>:       &lt;a href="mailto:a.haumer@haumer.at"&gt;A. Haumer&lt;/a&gt;
Modelica_UsersGuide_Conventions_UsersGuide.html: Invalid <a href>:         &lt;a href="http://www.haumer.at"&gt;Technical Consulting &amp;amp; Electrical Engineering&lt;/a&gt;&lt;br&gt;
Modelica_UsersGuide_Conventions_UsersGuide.html: Invalid <a href>:         &lt;a href="mailto:dr.christian.kral@gmail.com"&gt;C. Kral&lt;/a&gt;
Modelica_UsersGuide_Conventions_UsersGuide.html: Invalid <a href>:         &lt;a href="http://christiankral.net"&gt;Electric Machines, Drives and Systems&lt;/a&gt;&lt;br&gt;
Modelica_UsersGuide_ReleaseNotes.html: Invalid <a href>: modelica://Modelica.Media.Air.MoistAir.PsychrometricData
Modelica_Magnetic_FluxTubes_Interfaces.html: Invalid <img src>: <dl><dt>[image]</dt>
Modelica_Magnetic_QuasiStatic_FundamentalWave_Components.html: Invalid <img src>: ../Resources/Images/Magnetic/FundamentalWave/Components/vPhi
Modelica_Magnetic_QuasiStatic_FundamentalWave_Components.html: Invalid <img src>: ../Resources/Images/Magnetic/FundamentalWave/v_k
Modelica_Magnetic_QuasiStatic_FundamentalWave_Components.html: Invalid <img src>: ../Resources/Images/Magnetic/FundamentalWave/i_k

beutlich, 2017-08-21:
Did you run it on MSL 3.2.2+build.3 or current master HEAD? It sounds like a duplicate of #1991 to me if it was MSL 3.2.2+build.3.

d94pn, 2017-08-21:
Yes, MSL 3.2.2+build.3. Either close as duplicate or use the list as a checklist for any fix of #1991. The overlap of reported errors is probably not 100%.

beutlich, 2017-08-21:
* `Modelica_Math_Nonlinear.html: Invalid <a href>: ftp://ftp.inf.ethz.ch/doc/tech-reports/1998/306.ps`
was already reported in #2249 and fixed in master.
* The dead links to `https://www.fmi-standard.org` are a side-effect of modelica/fmi-standard.org#15.

beutlich, 2017-08-21:
Can you please run your checks on current master? Thank you.

closed 2017-08-22 by beutlich

d94pn, 2017-08-23:
After running on current master only theese remains according to curl:

Mechanics/Rotational.mo: Invalid <a href>: https://www.fmi-standard.org (should be http)
Mechanics/Rotational.mo: Invalid <a href>: https://www.fmi-standard.org

Various: 

http://christiankral.net (should be https)

But since web browsers redirects automatically I assume there is no need to be picky. There might even be a point in exposing e.g. http although it is really https, not sure.