modelica/ModelicaStandardLibrary #2307: Final modifier inheritance in Electrical.Analog sources

beutlich, 2017-08-22:
It is
```mo
model RampVoltage "Ramp voltage source"
  parameter SI.Voltage V(start=1) "Height of ramp";
  parameter SI.Time duration(min=Modelica.Constants.small, start=2) "Duration of ramp";
  extends Interfaces.VoltageSource(redeclare Modelica.Blocks.Sources.Ramp
    signalSource(final height=V, final duration=duration));
  // ...
end RampVoltage;
```
i.e. signalSource.height is redeclared with a final modifier. However, it is not the case in StepVoltage, e.g.,
```mo
model StepVoltage "Step voltage source"
  parameter SI.Voltage V(start=1) "Height of step";
  extends Interfaces.VoltageSource(redeclare Modelica.Blocks.Sources.Step
  signalSource(height=V));
  // ...
end StepVoltage;

Is there a special reason or is this just an inconsitency?

MartinOtter, 2017-08-23:
The correct way is to use "final" in a modifier for parameters. However, I guess that there are many places in MSL where final is not used.

christiankral, 2017-08-26:
The changes of a0d8187286dfdb6b2a0379e11b3e78e8e48b92db broke quite some models of the MSL. 

Example ``Modelica.Electrical.Analog.Semiconductors.HeatingDiode``:

The original implementation 
``extends Modelica.Electrical.Analog.Interfaces.ConditionalHeatPort(useHeatPort=true);``
was replaced by 
``extends Modelica.Electrical.Analog.Interfaces.ConditionalHeatPort(final useHeatPort=true);``, so the heat port is always on  -- which is not the intention of the model.

When running the test example ``Modelica.Electrical.Analog.Examples.HeatingRectifier`` then Dymola complains that the final parameter ``heatPort`` gets overwritten again by
`` Modelica.Electrical.Analog.Semiconductors.HeatingDiode HeatingDiode1(
      useHeatPort=true) ...;``
which is definitely not legal. 

I founrd some more broken models:
- Modelica.Electrical.Machines.Examples
- Modelica.Magnetic.FundamentalWave.Examples.BasicMachines

beutlich, 2017-08-26:
Do you recommend to revert a0d8187286dfdb6b2a0379e11b3e78e8e48b92db because of the trouble with useHeatPort?

Furthermore, there are some ConditionalHeatPort that take a parametric T_Ref and other that take a literal temperature of 293.15K.

beutlich, 2017-08-26:
OK, reverted by 501a04def2154c5c4a981596fa2d86f9ce6cc19a.

christiankral, 2017-08-26:
OK, I think this is good decision for now. Thanks.

beutlich, 2017-08-28:
New commit 970c5b556c770279698536217211c3f5a5ac4a94 kept the useHeatPort modifications untouched.

kristinmajetta, 2017-08-29:
Having a look into the model Modelica.Electrical.Analog.Sources.StepVoltage signalSource.height IS redeclared with a final modifier. Are we talking about the same model here?
```
model StepVoltage "Step voltage source"
  parameter SI.Voltage V(start=1) "Height of step";
  extends Interfaces.VoltageSource(redeclare Modelica.Blocks.Sources.Step
      signalSource(final height=V));
...
end StepVoltage
```

tbeu, 2017-08-29:
>  Having a look into the model Modelica.Electrical.Analog.Sources.StepVoltage signalSource.height IS redeclared with a final modifier. Are we talking about the same model here?

Yes, after 970c5b556c770279698536217211c3f5a5ac4a94 it is.

closed 2017-09-07 by beutlich