modelica/ModelicaStandardLibrary #2309: Add variable for representing torque/force in electromechanic converters

christiankral, 2017-08-23:
In the models

- Modelica.Electrical.Analog.Basic.EMF
- Modelica.Electrical.Analog.Basic.TranslationalEMF

there is, unfortunately, no alias variable representing torque and force, respecitively.

In order to unify the electromechanical components including machines, it makes sense to additionally introduce the following variables:
  
- Modelica.Electrical.Analog.Basic.EMF
  ```modelica
   tau = flange.tau;
   tauElectrical = -flange.tau;
  ```
- Modelica.Electrical.Analog.Basic.TranslationalEMF
  ```modelica
   f = flange.f;
   fElectrical = -flange.f;
  ```

beutlich, 2017-08-23:
tauElectrical / fElectrical may then also be used in the internalSupport modification for f / tau.

christiankral, 2017-08-23:
I probably wouldn't do that to better maintain the symmetry of the torques of the flange and the internal support, as the electrical torque and force rather represent alias variables.

In case of Modelica.Electrical.Analog.Basic.EMF I would go for the following equations (an keep the rest as it is):

```modelica
  SI.Torque tau "Torque of shaft";
  SI.Torque tauElectrical "Electrical torque";
equation
  tau = flange.tau;
  tauElectrical = - tau;
```

christiankral, 2017-08-23:
Actually, I would additionally change the original implementation to
```modelica
  Mechanics.Rotational.Interfaces.InternalSupport internalSupport(tau=-tau)
equation
  tau = -k*i;
```

beutlich, 2017-08-23:
This is what I meant.
```mo
Mechanics.Rotational.Interfaces.InternalSupport internalSupport(tau=tauElectrical);
```

closed 2017-08-23 by christiankral