modelica/ModelicaStandardLibrary #2310: Modelica C-Sources on dSpace

chria, 2017-08-25:
I have a question regarding the Modelica C-Sources (located in MSL/Resources/C-Sources). For the release of MSL 3.2.1 the C-Sources was tested and verified to work on dSpace DS1006 (from the release notes). I'm wondering if the same tests was performed for MSL 3.2.2? 

I'm asking because I'm having trouble compiling the C-Sources of ModelicaInternal.c and ModelicaRandom.c on DS1006 (even with NO_FILE_SYSTEM define). Has anyone tested to compile these files to work on together with DS1006? If so, is there anything more that needs to be set than the mentioned define?

Thanks!

beutlich, 2017-08-25:
What dSPACE Release do you use?

sjoelund, 2017-08-25:
ModelicaInternal.c I would assume should still compile. ModelicaRandom.c is a new file only used for the randomization routines. I guess it's a bit unfortunate it's part of the ModelicaExternalC blob since it can't then be easily skipped if the target doesn't support it... Is it a problem that you need to set NO_TIME as well?

beutlich, 2017-08-25:
Yes, NO_TIME and NO_PID need to be defined.

chria, 2017-08-25:
Regarding which version of dSpace I'm using, good question. I see that I use the compiler from the folder "dSpace RCPHIL 2013-A". Does that help?

The error that I'm seeing is first in ModelicaInternal.c, where the problem is related to the check for POSIX,

#if defined(__unix__) || defined(__linux__) || defined(__APPLE_CC__)
  #include <unistd.h>
#endif

#if !defined(_POSIX_) && defined(_POSIX_VERSION)
  #define _POSIX_ 1
#endif

In MSL 3.2.1 this check was inside the NO_FILE_SYSTEM define but was moved outside in MSL 3.2.2 and now cause problem (at least for me).

For the ModelicaRandom.c, the error that I'm seeing (with the defines NO_FILE_SYSTEM, NO_TIME, NO_PID) is show below:
...
ModelicaRandom.c:123: error: `PTHREAD_MUTEX_INITIALIZER' undeclared here (not in a function) 
ModelicaRandom.c: In function `ModelicaRandom_setInternalState_xorshift1024star': 
ModelicaRandom.c:432: warning: implicit declaration of function `pthread_mutex_lock' 
ModelicaRandom.c:440: warning: implicit declaration of function `pthread_mutex_unlock'

chria, 2017-08-25:
Great, thanks!

beutlich, 2017-08-25:
Resolved in master by cf7fcb8fe0f972d99455043fcfa8982b358e2ed3. Besides NO_PID and NO_TIME you need to define _SSIZE_T to avoid the inclusion problem with unistd.h and the new one NO_MUTEX.

closed 2017-08-25 by beutlich

beutlich, 2017-08-25:
P.S.: ModelicaStrings.c will give linker errors in dSPACE Release 2013-A due to unresolved symbols tolower, toupper, isspace, isdigit and isalpha. This is fixed in [dSPACE Release 2013-A SP3](https://www.dspace.com/de/gmb/home/support/patches/suppatchsfc/rls2013aimplswspx.cfm):
```
DS1006 Compiler 2.2p1:
  - This version contains a bugfix for a linker error that occurs as soon
    as a function from the <ctypes.h> header is used.
```

chria, 2017-08-25:
Thanks! I'll check it out.