modelica/ModelicaStandardLibrary #2311: Remove "import Modelica;"

christiankral, 2017-08-28:
The following files contain ``import Modelica;`` in the Modelica Standard Library.
```
./Magnetic/FundamentalWave.mo
./Magnetic/QuasiStatic/FundamentalWave.mo
./Mechanics/Rotational.mo
./Mechanics/Translational.mo
./Mechanics/MultiBody/Parts.mo
./Mechanics/MultiBody/Frames.mo
./Electrical/PowerConverters.mo
./Electrical/QuasiStationary/MultiPhase.mo
./Electrical/MultiPhase.mo
./Electrical/Machines.mo
./Utilities/Strings.mo
./Fluid/Utilities.mo
./Fluid/Dissipation.mo
./Blocks/Math.mo
./Blocks/package.mo
./Blocks/Continuous.mo
./Math/Nonlinear.mo
./Math/package.mo
./Media/package.mo
```

On my opinion these imports shall be removed since they do not serve any purpose.

HansOlsson, 2017-08-28:
Not all of them are meaningless.  It seems correct for Modelica.Magnetic and Modelica.Electrical, but in Modelica.Mechanics the following are encapsulated 
Modelica.Mechanics.MultiBody.Parts.BevelGear1D.Housing (and similar ones) Modelica.Mechanics.MultiBody.Frames.Orientation.equalityConstraint  (and similar ones) Modelica.Mechanics.Rotational.Components.InitializeFlange.Set_phi_start (and similar ones)
(It seems likely this cover all the cases - and similarly in Modelica.Utilities and Modelica.Fluid)

In an encapsulated class it is necessary to either use ".Modelica" or "import Modelica" when referring to Modelica.

beutlich, 2017-08-28:
Resolved by 3b7fd5e2118cf1986af069fb0633fccc156ef1d6 in master.

beutlich, 2018-02-26:
3fd3d6a736386e9c59a13c61559b38293d62d929 reintroduced the supefluous import statement.