modelica/ModelicaStandardLibrary #2312: Redundant model ConditionalHeatPort

beutlich, 2017-08-28:
Modelica.Magnetic.FluxTubes.Interfaces.ConditionalHeatPort is identical to Modelica.Thermal.HeatTransfer.Interfaces.PartialElementaryConditionalHeatPort and should be set obsolete for now and removed later.

beutlich, 2017-10-05:
Duplicate of #359 (7.5 years ago, sigh), but scheduled there for next major release.

christiankral, 2017-12-31:
The three partial heat port models in the packages ```FluxTubes```, ```Analog``` and ```HeatTransfer``` are not compatible as they are using different variable names:

Variables | ```FluxTubes``` | ```Analog``` |  ```HeatTransfer```
--------------- | ------------------- | ---------------- | ------------------------
loss power | ```LossPower``` | ```LossPower ``` | ``` lossPower ```
temperature of heat port  | ```T_heatPort``` | ```T_heatPort``` | ```TheatPort```

Therefore I propose to switch the milestone to the next major release where conversion scripts are allowed. I propose to keep the partial model in ```HeatTransfer``` and remove the other other two partial models in the next major release. In this case we could even consider to maintain the name ```T_heatPort``` instead of ```TheatPort```. 

In addition to the conditional heat port models, the vectorized version of ```Modelica.Electrical.MultiPhase.Interfaces.ConditionalHeatPort``` should also go to the sub-package ```Modelica.Thermal.HeatTransfer.Interfaces```. Then we have everything in one place.

I am not sure: shall we still mark the partial models in ```FluxTubes``` and ```Analog``` as obsolete, as they are not compatible to the partial model in ```HeatTransfer```?

christiankral, 2017-12-31:
One more thing to note is the different location of the heat port of the icon. 

Appearance | ```FluxTubes``` | ```Analog``` |  ```HeatTransfer```
--------------- | ------------------- | ---------------- | ------------------------
location of heat port | lower center | lower center | lower left

Unifying the three models may cause some models to have misplaced heat ports.

closed 2018-01-02 by beutlich