modelica/ModelicaStandardLibrary #2313: Placement annotation with zero width and zero height extent

beutlich, 2017-08-28:
The Real output connectors phi_motor and w of Modelica.Blocks.Examples.NoiseExamples.Utilities.Parts.MotorWithCurrentControl both have a zero width and zero height extent in the icon layer with the intention to set them invisible.
```mo
Modelica.Blocks.Interfaces.RealOutput phi_motor(unit="rad", displayUnit="deg")
"Rotational position"
annotation (Placement(transformation(extent={{60,20},{80,40}}),
    iconTransformation(extent={{90,40},{90,40}})));
Modelica.Blocks.Interfaces.RealOutput w(unit="rad/s") "Rotational speed"
annotation (Placement(transformation(extent={{60,0},{80,20}}),
    iconTransformation(extent={{90,40},{90,40}})));
```
However, this is neither correctly specified nor correctly handled by all tools.

@HansOlsson Is there a clean way to set a connector invisible in either icon layer xor diagram layer?

HansOlsson, 2017-08-30:
> Is there a clean way to set a connector invisible in either icon layer xor diagram layer?

Currently not:

-  There is a clean way to make a connector invisible in both (visible=false), but not differently for icon and diagram.
-  It is possible disable drawing of inherited primitives on a per layer basis; but not connectors/components as in this case.

beutlich, 2017-08-30:
My idea is to introduce new iconVisible Boolean flag in a backward compatible way. If it is not set, it falls back to visible, otherwise iconVisible is the flag for the icon layer, and visible for the diagram layer only. Should I make a Trac proposal?

HansOlsson, 2017-08-30:
> My idea is to introduce new iconVisible Boolean flag in a backward compatible way. If it is not set, it falls back to visible, otherwise iconVisible is the flag for the icon layer, and visible for the diagram layer only.  
> Should I make a Trac proposal?

Sure. 
However, there are other option - in particular it would be more orthogonal for me to have visible inside the transformation; so that you can either hide in icon or diagram. Obviously one would have to clarify if visible for the icon overrides the component-visible (I believe that to be preferable) or both must be true.

beutlich, 2017-09-13:
>> Should I make a Trac proposal?

> Sure.

Done. Please see https://trac.modelica.org/Modelica/ticket/2200.

svorkoetter, 2018-06-14:
Continuing the discussion from #2630 ... MapleSim currently renders it like this:
[image]
Notice the tiny dot at the location of the two "invisible" connectors. Even though they have mathematical size zero, the thickness of the "pen" used to draw them is not zero, so they are still visible. Also, it is possible to connect to them in the GUI (although it is hard to choose the right one, since they are coincident). Is there any reason that these need to be connectors at all?

svorkoetter, 2018-06-14:
Note also that, at least in MapleSim, these connectors end up listed in the help as connectors of the component.

tobolar, 2018-06-14:
Isn't this a Modelica Specification issue?

It is not unusual for me to use such constructs to graphically highlight some variables. Is there any other option?

HansOlsson, 2018-06-14:
Yes, it is specification issue specifically: https://trac.modelica.org/Modelica/ticket/2200

However, that doesn't explain what to do for this release of MSL.

tobolar, 2018-06-14:
The variables/connectors are apparently not used in the example `Blocks.Examples.NoiseExamples.ActuatorWithNoise`, so we could delete it without any concern.

PR #3385 (2020-02-03): Change "invisible" output connectors to output Reals in Modelica.Blocks.Examples.NoiseExamples.Utilities.Parts.MotorWithCurrentControl
This resolves #2313 as proposed. No conversion is required for examples.
closed 2020-02-03 by beutlich