modelica/ModelicaStandardLibrary #2317: Non-existent class referenced in choices annotation in Modelica.Mechanics.MultiBody.Examples.Loops.Utilities.EngineV6_analytic

harmanpa, 2017-09-04:
In Modelica.Mechanics.MultiBody.Examples.Loops.Utilities.EngineV6_analytic there is a choices annotation for redeclaring the Cylinder type. One of the options is "Modelica.Mechanics.MultiBody.Examples.Loops.Utilities.Cylinder_analytic", however no such class exists.

https://github.com/modelica/Modelica/blob/f1e1cded603cf9dc8ac8f48da4fa61cecf6df129/Modelica/Mechanics/MultiBody/Examples/Loops/Utilities.mo#L746

HansOlsson, 2017-09-04:
The correction is not clear.
It is not clear why this model Modelica.Mechanics.MultiBody.Examples.Loops.Utilities.Cylinder_analytic would be needed in addition to Modelica.Mechanics.MultiBody.Examples.Loops.Utilities.Cylinder_analytic_CAD

But without Cylinder_analytic it seems odd to have a choice-annotation with only one choice - which is the default.

So one possibility is to just remove the entire choices-part.

beutlich, 2017-10-15:
c24c607b0e1591abce46b84e734e39aa6e7d2b9a was the commit that uncommented the choices annotation. (10 years ago). Telling from the commit message it is not clear if it was by intention or not.
Since the CylinderBase model also contains CAD part visualization, I agreed to comment the choices annotation again.

PR #2368 (2018-01-08): Remove choices annotation for Cylinder model
Cylinder_analytic does not exist and having a single choice with Cylinder_analytic_CAD is not a real choice.

Close #2317.
closed 2018-01-08 by tobolar