modelica/ModelicaStandardLibrary #2320: Implement License change decision to change to BSD 3-clause in MSL

hubertus65, 2017-09-07:


beutlich, 2017-09-08:
Can we please also decide on all copyright disclaimers in Modelica files and C-sources?! As we discussed it in Praha it turned out the current copyright disclaimers are invalid w.r.t. the MA.

hubertus65, 2017-09-11:
You mean copyright disclaimers claiming copyright by the MA, since we do not have a contributors license agreement?

beutlich, 2017-09-11:
Exactely. The ones like [here](https://github.com/modelica/Modelica/blob/master/Modelica/Resources/C-Sources/ModelicaStandardTables.c#L3).

hubertus65, 2017-09-11:
Hmmm. I am not certain about that. If anyone legally worked as a private person with this, and the person itself would say this was done on behalf of the MA, it would be correct. This certainly holds for academics in a number of countries. I would not touch this and rather spend effort on the CLA, which would make it correct.

beutlich, 2017-09-13:
@modelica/map-lib-write I created the new branch issue2320-license (based on master) where all of the above subtasks should go before they will be merged.

hubertus65, 2017-09-15:
What do you mean by "Remove classes that declare ModelicaLicense2 (and accept breaking backward compatibility)"? Do you think there are classes that were intentionally put under Modelica LIcense2 and can't be changed to BSD?

beutlich, 2017-09-18:
> What do you mean by "Remove classes that declare ModelicaLicense2 (and accept breaking backward compatibility)"? 

My intention was to completely remove the classes Modelica.UsersGuide.ModelicaLicense2 and ModelicaServices.UsersGuide.ModelicaLicense2 (instead of setting it obsolete). Since these classes are documentation-only classes, the case of breaking the backward compatibility should be acceptable. Let's see what the project manager @AHaumer thinks about it.

dietmarw, 2017-10-07:
I agree that removing the licence classes is an acceptable non-backward compatible change. I can not think any models *depending* on it.

beutlich, 2017-10-09:
> I agree that removing the licence classes is an acceptable non-backward compatible change. 

You are right. Since Modelica.Media.UsersGuide.Future was removed by 42e435633eb5b642977a8ea2468ba3de226917cb recently, we also are allowed to remove the classes Modelica.UsersGuide.ModelicaLicense2 and ModelicaServices.UsersGuide.ModelicaLicense2.

hubertus65, 2017-10-09:
My co-worker Jesse Gohl is going to work on this issue and send you a pull request. /Hubertus

beutlich, 2017-10-09:
@jbgohl Make sure to your pull request targets the issue2320-license feature branch. Thanks.

jbgohl, 2017-10-16:
Regarding the task "Change license comments to BSD-3-Clause in Modelica files", the current wording in the Modelica files is of the form:

> For license conditions (including the disclaimer of warranty) see...

Included in this text are links to both the ModelicaLicense2 class and the license text on the modelica.org website at https://www.modelica.org/licenses/ModelicaLicense2. 

Are we considering changing this above text in all the Modelica files to the full BSD 3-clause text, including the 3 clauses and disclaimer?

If not, do we want to still include a "License" class in the User's Guide package that lists the license text that can be viewed within Modelica editors? Or is it sufficient to simply point to the website https://opensource.org/licenses/BSD-3-Clause?

Just looking for a little clarification on what is meant by "Change license comments..."

beutlich, 2017-10-16:
I give an example of what I meant by "Change license comments to BSD-3-Clause in Modelica files". There are license disclaimers of the kind in many Modelica files. 
https://github.com/modelica/Modelica/blob/b50fc360c3113b42bd872222149c697678e8fa1d/Modelica/Fluid/Dissipation.mo#L13057-L13064
This needs to be adapated to something like
```
<p>
<strong>Licensed by the Modelica Association under the 3-Clause BSD License</strong><br>
Copyright &copy; 2007-2016, XRG Simulation GmbH.
</p>
<p>
<em>This Modelica package is <u>free</u> software and the use is completely at <u>your own risk</u>; it can be redistributed and/or modified under the terms of the 3-Clause BSD License. For license conditions (including the disclaimer of warranty) visit <a href=\"https://opensource.org/licenses/BSD-3-Clause\">https://opensource.org/licenses/BSD-3-Clause</a>.</em>
</p>
```
i.e., just the link.

jbgohl, 2017-10-16:
Ok, that only leaves ticket #2285 and the release notes.  Regarding #2285, who owns the copyright and what are the dates?  The first commit I found was in 2010 by Martin but that is in Git, not sure if it existed earlier in Subversion. Then regarding the release notes, how should they be provided?

beutlich, 2017-10-16:
@jbgohl Good job! I removed the #2285 and release notes task again, which can be considered as separate jobs.

beutlich, 2017-11-08:
I deleted the issue2320-license feature branch again since I opened #2373 from my own repo.

PR #2373 (2018-01-19): Merge issue2320-license aka Change to 3-Clause BSD License
Thanks to substantial work of @jbgohl the feature branch issue2320-license is able to get merged to finally close #2320. See also #2372 for a discussion.

@modelica/map-lib-write @adeas31 @adrpo @AHaumer @andreas-junghanns @bernhard-thiele @casella @christiankral @d-hedberg @DagBruck @dietmarw @ECClauss @GallLeo @HansOlsson @harmanpa @hubertus65 @jespermattsson @kristinmajetta @MartinOtter @mtiller @mwetter @rfranke @sjoelund @svorkoetter @ThomasBoedrich @wischhusen @ziske Please comment and give your OK as library officer or contribut [... cut]
closed 2018-01-19 by hubertus65