modelica/ModelicaStandardLibrary #2326: ExternalCombiTable1D constructor always displays the filename on load

adrpo, 2017-09-25:
This seems to have changed from MSL3.2.2 when no filename was displayed. Is this intentional or a bug?

ExternalCombiTable1D calls ModelicaStandardTables_CombiTable1D_init which calls ModelicaStandardTables_CombiTable1D_init2 with verbose set to 1. 
I think it should be set to zero instead of one to have the same behavior as before.

beutlich, 2017-09-25:
The verbose flag moved from the read functions to the new init2 constructors (since the tables are read in the constructors due to #1899). For the old init constructors (that is using MSL 3.2.2 with latests C-Sources) the default verbose=1 is used, without any chance to configure this behaviour.  Is this acceptable for you?

adrpo, 2017-10-02:
Yeah, is OK, it impacted one of our tests, but I found a way to workaround it.

adrpo, 2017-10-02:
You can probably close this issue as I guess other tool providers are not affected by this.

beutlich, 2017-10-02:
The issue always cccurs if a tool vendor provides master ModelicaStandardTables and user loads MSL v3.2.1 or v3.2.2. So we need to live with it.

closed 2017-10-02 by beutlich

sjoelund, 2017-10-02:
There could be an environment variable to allow the tool to set the behavior, but I guess it's mostly just relevant if the exact output of the executable needs to be the same as before...

beutlich, 2017-10-02:
> There could be an environment variable to allow the tool to set the behavior, but I guess it's mostly just relevant if the exact output of the executable needs to be the same as before...

Yes and yes.