modelica/ModelicaStandardLibrary #2327: Newly introduced transformer components of FluxTubes use entirely different variable naming

christiankral, 2017-09-25:
There are newly introduced components in the FluxTubes package related with hysteresis. I updated the the default component names according to #2201 and changed the component names in the newly introduced examples, since they have NOT to be backwards compatible, yet.

However, since we already have transformer models in the MSL I would prefer to have the newly introduced transformer models to have parameters mostly equal to the existing ones. Since I put a lot of effort in unifying the appearance and the usability of the electrical and magnetic sub packages of the MSL I am kindly asking @ThomasBoedrich to unify the naming of the transformer parameters in 

- Modelica.Magnetic.FluxTubes.Examples.Hysteresis.Components.Transformer1PhaseWithHysteresis 
- Modelica.Magnetic.FluxTubes.Examples.Hysteresis.Components.Transformer3PhaseYyWithHysteresis

to better match the models of 

- Modelica.Electrical.Machines.BasicMachines.Transformers
- Modelica.Electrical.QuasiStationary.Machines.BasicMachines.Transformers

If you ask me to, I can make the changes. 

One more note: I already renamed the instance names in 

- Modelica.Magnetic.FluxTubes.Examples.Hysteresis
- Modelica.Magnetic.FluxTubes.Examples.Hysteresis.Components

as they shall rather start with a lower case than an upper case letter. The renaming of the parameters is an open issue as I did not want to make a change without giving notice. (This probably needs a clearer advice in Modelica.UsersGuide.Conventions.)

ThomasBoedrich, 2017-09-29:
Dear Christian,

many thanks for your effort to unify the naming of the parameters in the different transformer models of the MSL. 

As offered by you I hereby kindly ask you to make the name changes as suggested.  

Concerning the start of all instance names with a lower-case letter:
I would prefer upper-case instance names that are established in electrical engineering, i.e. an upper-case R for a resistor, L for an inductor and C for a capacitor. These common names (with upper-case) are used for example in the instance names of Modelica.Electrical.Analog.Examples. Moreover, a lower-case r commonly denotes a differential resistance. However, if you consider it necessary or advisable to change the instance names in the mentioned examples of Modelica.Magnetic.FluxTubes, than go ahead with these changes, too. 

Best regards
Thomas

christiankral, 2017-12-30:
The parameter names are now unified to better match the models of ```Modelica.Electrical.Machines.Transformers```. The main changes applied to the indexes:

- ```_p``` --> ```1``` for primary quantities
- ```_s``` --> ```2``` for secondary quantities

However, I did not change the structure of the models. Therefore, the remaining differences between the three phase transformer model of the ```FluxTubes``` and the ```Machines``` packages are:

Implementations | ```FluxTubes``` | ```Machines``` 
---------------------- | -------------------- | ------------------
Primary electrical connector(s) | three pins | one plug
Secondary electrical connector(s) | three pins | one plug
Thermal connector(s) | heat port (1 x total loss) | heat port (2 x winding and 1 x core losses)
Operating temperature(s) for windings and core | not separated | separated 

@ThomasBoedrich please check the applied modifications and close the ticket if you are OK with the changes.

ThomasBoedrich, 2018-01-16:
Dear Christian,

thank you very much for the gentle name changes. 

Best regards
Thomas

closed 2018-01-16 by ThomasBoedrich