modelica/ModelicaStandardLibrary #2330: Large number of new models introduced in maint/3.2.2

sjoelund, 2017-09-29:
It seems recently a large number of new (example) models have been introduced to maint/3.2.2 without adding corresponding reference files to the regression suite. Were these added by mistake or on purpose? Either way we need reference files for the models since they have been in the master branch for some time now.

The following files are currently missing (or perhaps a new tag should rather be created in the regression suite):
```
FundamentalWave/Examples/BasicMachines/AIMC_Conveyor/AIMC_Conveyor.mat
FundamentalWave/Examples/BasicMachines/AIMC_Initialize/AIMC_Initialize.mat
FundamentalWave/Examples/BasicMachines/AIMC_Inverter/AIMC_Inverter.mat
FundamentalWave/Examples/BasicMachines/AIMC_Steinmetz/AIMC_Steinmetz.mat
FundamentalWave/Examples/BasicMachines/AIMC_Transformer/AIMC_Transformer.mat
FundamentalWave/Examples/BasicMachines/AIMC_withLosses/AIMC_withLosses.mat
FundamentalWave/Examples/BasicMachines/AIMC_YD/AIMC_YD.mat
FundamentalWave/Examples/BasicMachines/SMEE_LoadDump/SMEE_LoadDump.mat
FundamentalWave/Examples/BasicMachines/SMEE_Rectifier/SMEE_Rectifier.mat
FundamentalWave/Examples/BasicMachines/SMPM_Braking/SMPM_Braking.mat
FundamentalWave/Examples/BasicMachines/SMPM_CurrentSource/SMPM_CurrentSource.mat
FundamentalWave/Examples/BasicMachines/SMPM_VoltageSource/SMPM_VoltageSource.mat
QuasiStatic/FundamentalWave/Examples/BasicMachines/InductionMachines/IMC_Characteristics/IMC_Characteristics.mat
QuasiStatic/FundamentalWave/Examples/BasicMachines/InductionMachines/IMC_Conveyor/IMC_Conveyor.mat
QuasiStatic/FundamentalWave/Examples/BasicMachines/InductionMachines/IMC_Initialize/IMC_Initialize.mat
QuasiStatic/FundamentalWave/Examples/BasicMachines/InductionMachines/IMC_Transformer/IMC_Transformer.mat
QuasiStatic/FundamentalWave/Examples/BasicMachines/InductionMachines/IMC_withLosses/IMC_withLosses.mat
QuasiStatic/FundamentalWave/Examples/BasicMachines/InductionMachines/IMC_YD/IMC_YD.mat
QuasiStatic/FundamentalWave/Examples/BasicMachines/InductionMachines/IMS_Characteristics/IMS_Characteristics.mat
```

sjoelund, 2017-09-30:
The same batch of changes also had a negative impact on simulation performance (at least in OpenModelica): https://libraries.openmodelica.org/branches/history/master/2017-09-29%2004:39:52..2017-09-30%2012:33:00.html (a few models in that test, such as the buildings model are on the limit of triggering the timeout and only sometimes with, but there are many of Christian's models affected by the changes and I guess there shouldn't be big changes in the maintenance branch)

christiankral, 2017-10-08:
The new models are added by purpose, see #2276. I will be adding the reference files as soon as the developments are finalized. 

I need to check on the performance issues, but however, the models of Modelica.Electrical.Machines have not been changed...

beutlich, 2018-11-12:
@christiankral Is there anything left to do here or can we close this issue as wontfix?

christiankral, 2018-11-16:
I created a variables list on May 15, 2018 and sent to to @GallLeo. Reference cases should therefore exist.

closed 2018-11-16 by christiankral