modelica/ModelicaStandardLibrary #2340: Modelica.Fluid.Dissipation vs. FluidDissipation

beutlich, 2017-10-04:
Modelica.Fluid.Dissipation was added to MSL v3.2 in March 2010 (at the time the EuroSysLib project was almost finished). There also is [FluidDissipation](https://github.com/ClaRaLibrary/FluidDissipation) from [XRG Simulation](http://xrg-simulation.de/produkte/xrg-library/xrg-fluiddissipation-library) and it turns out that FluidDissipation and Fluid.Dissipation are out of synchronized maintenance. For instance, #2213 and #2336 were already fixed in FluidDissipation but where missing in Fluid.Dissipation when discovered there. Furthermore, FluidDissipation has many new components (for example, PressureLoss.Diffuser), which are missing in Fluid.Dissipation.

Since we do not have a modularized MSL it would be good to know if it is intended that bug fixes on FluidDissipation are ported to Fluid.Dissipation and vice versa. @wischhusen Can you please comment on the maintenance strategy. Thanks a lot! (Open comment from https://github.com/modelica/Modelica/pull/2213#issuecomment-288184934).

wischhusen, 2018-01-04:
Modelica.Fluid.Dissipation is a subset of the FluidDissipation library which was decided when the library was integrated into MSL. XRG has done maintenance for the same issues in both libraries. We don't see a problem with correcting failures in both versions.

beutlich, 2018-01-04:
> We don't see a problem with correcting failures in both versions.

That is nice to read - at least for the maintenance part of the question. However, practice prooved wrong when stumbling on #2213 and #2336.

wischhusen, 2018-01-04:
Thanks for this notification! Please note, that someone else did the changes. As soon as those changes are made I do not see this issue anymore in my list. Unfortunately, I am not able to scan the list day by day. The general problem is, that there are times were I am not able to filter 100 github notification mails per month.

In order to react faster and also be able to be aware of changes other people perform in my packages I would like to have only notifications on those packages I am responsible for. Can this be somehow configured? Thanks for your help.

beutlich, 2018-01-04:
You can check the issues you have been assigned to: https://github.com/modelica/Modelica/issues?utf8=%E2%9C%93&q=is%3Aopen+assignee%3Awischhusen+

It also helps to remove your assignment from issues you are not interested in and to stop watching the repo. You still get notified for issues you will be assigned to or you previously described.

wischhusen, 2018-01-04:
I will try. Thank you!

closed 2018-11-09 by beutlich