modelica/ModelicaStandardLibrary #2343: Internal variables of Modelica.Electrical.Analog.Semiconductors.HeatingNPN are public

beutlich, 2017-10-05:
The internal variables of Modelica.Electrical.Analog.Semiconductors.HeatingNPN are public and miss a description string. In comparison, similar internal variables are protected in Modelica.Electrical.Analog.Semiconductors.HeatingPNP. This looks neither consistent nor professional.

[image]

See #2268, where this issue originates from.

beutlich, 2017-10-06:
These public variables were introduced by 6a90d1a8931b29d0d60547f5b3329191ecfe2bfd (for MSL v1.6), thus, from the very first beginning.

christiankral, 2017-10-06:
Most ```Real``` variables shall also have a physical unit, as they represent physical quantities like
- ```vbc = B.v - C.v``` is a voltage 
- ```is_t = Is*pow((T_heatPort/Tnom), XTI)*htempexp``` is a current

This issue also applies to the variables of quite some other models of ```Modelica.Electrical.Analog.Semiconductors```

beutlich, 2017-10-06:
The only reason I can think of why HeatingNPN has public internal variables, is the existing sample model Modelica.Electrical.Analog.Examples.HeatingNPN_OrGate, while there is neither an example nor test model for HeatingPNP. Setting the internal variables of HeatingNPN to protected resulted in #2268. For me, it therefore looks, that the design of HeatingNPN and HeatingPNP is rather questionable.

kristinmajetta, 2018-05-29:
- added description strings and units to mentioned variables in Modelica.Electrical.Analog.Seminconductors.HeatingNPN, Modelica.Electrical.Analog.Seminconductors.HeatingPNP,
Modelica.Electrical.Analog.Seminconductors.NPN, Modelica.Electrical.Analog.Seminconductors.PNP
- made mentioned variables public in model  Modelica.Electrical.Analog.Seminconductors.HeatingPNP to  make the model consistent to the other three models

SHA-1: 2766383cc656a74b4c20fb8b756364354f48be5e
SHA-1: 5eaf686e024d61533d1095b421fbaeddb084f385
SHA-1: aa6b3d0b66c5bad8d15d86981b691ec34c941618
SHA-1: 68f6be7ed9dd65c8fffe7e4e0d85c58e69f37640

dietmarw, 2018-05-29:
Reopening so that the case comments can be corrected.

closed 2018-05-29 by beutlich