modelica/ModelicaStandardLibrary #2346: Remove transparency from group images

beutlich, 2017-10-09:
When discussing #2344, it was agreed, that the PNG images used as group images should have white background. This refers to the following images in Magnetic.FluxtTubes

- [ ] CoaxCylindersEndFaces.png
- [ ] Core_SinglePhase.png
- [ ] Core_ThreePhase1.png
- [ ] CuboidOrthogonalFlux.png
- [ ] CuboidParallelFlux.png
- [ ] EighthOfHollowSphere.png
- [ ] EighthOfSphere.png
- [ ] Eqn_ShapeFunctionsTellinen.png
- [ ] HalfCylinder.png
- [ ] HalfHollowCylinder.png
- [ ] HardMagneticHysteresis.png
- [ ] HollowCylinderAxialFlux.png
- [ ] HollowCylinderRadialFlux.png
- [ ] LeakageAroundPoles.png
- [ ] LeakageWithCoefficient.png
- [ ] MovingCoilActuator_dimensions.png
- [ ] QuarterCylinder.png
- [ ] QuarterHollowCylinder.png
- [ ] QuarterHollowSphere.png
- [ ] QuarterSphere.png
- [ ] SoftMagneticHysteresis1.png
- [ ] Solenoid_dimensions.png

HansOlsson, 2017-10-09:
I thought the agreement was to first discuss transparency for images and then possibly introduce it consistently, instead of introducing "randomly" combined with other changes.

However, the above seems like existing images - and I don't see a reason to spend time on them before that discussion.

MartinOtter, 2017-10-09:
Yes. First discuss, before making any changes to images

beutlich, 2017-10-09:
Thought we wanted to have white background already now - everywhere.

HansOlsson, 2018-03-28:
I have added a Modelica specification ticket about this (proposing to standardize on transparent background - that is intended to be white-ish): https://trac.modelica.org/Modelica/ticket/2244

Note that the discussion vs. specification text-ratio is somewhat high for that proposal.

beutlich, 2018-05-29:
Reschudling milestone.