modelica/ModelicaStandardLibrary #2360: Missing test or example model for Modelica.Electrical.Analog.Semiconductors.HeatingPNP

beutlich, 2017-10-14:
Similar issue as #2083, but derived from #2343.

Whereas there is one example model utilizing Semiconductors.HeatingNPN (in Examples.HeatingNPN_OrGate) there is no such model for Semiconductors.HeatingPNP in MSL or ModelicaTest.

kristinmajetta, 2018-05-02:
added testexample Modelica.Electrical.Analog.Examples.HeatingPNP_NORGate 
SHA-1: 93239b87b26e5f0b781c5a69c86799dee74e335b

beutlich, 2018-05-14:
Modelica.Electrical.Analog.Examples.HeatingPNP_NORGate is illegal Modelica since the protected component vt_t is modified. See #2343 (especially https://github.com/modelica/ModelicaStandardLibrary/issues/2343#issuecomment-334679975) for the dilemma of public internal variables vs. modified protected variables in both HeatingPNP and HeatingNPN.

kristinmajetta, 2018-05-29:
Since the model Modelica.Electrical.Analog.Semiconductors.HeatingPNP has no longer internal variables, the testmodel Modelica.Electrical.Analog.Examples.HeatingPNP_NORGate should be legal now.

closed 2018-05-29 by kristinmajetta