modelica/ModelicaStandardLibrary #2370: Error after simulating Modelica.Fluid.Examples.HeatingSystem in OMEdit

DarcyCorbett, 2017-10-16:
After I try simulating the heating system example, I get the following error in the OM Edit output dialog box:

stdout | OMEditInfo | <p>C:/Users/Darcy/OneDrive/Documents/Masters/Cyprus/Eclipse/EclipseLocalRepo/Pentakomo/PentakomoModelling/Working Directory/Modelica.Fluid.Examples.HeatingSystem.exe -port=55016 -logFormat=xmltcp -override=startTime=0,stopTime=6000,stepSize=12,tolerance=1e-06,solver=dassl,outputFormat=mat,variableFilter=.* -r=Modelica.Fluid.Examples.HeatingSystem_res.mat -jacobian=coloredNumerical -w -lv=LOG_STATS</p>
assert | warning | <p>The following assertion has been violated at time 0.000000<br>
pump.delta_head_init &gt;= 0.0</p>
assert | warning | <p>Variable violating min constraint: 0.0 &lt;= pump.delta_head_init, has value: -2.0431</p>
stdout | warning | <p>While solving non-linear system an assertion failed during initialization.</p>
stdout | warning | <p>The non-linear solver tries to solve the problem that could take some time.</p>
stdout | warning | <p>It could help to provide better start-values for the iteration variables.</p>
stdout | warning | <p>For more information simulate with -lv LOG_NLS</p>
stdout | warning | <p>While solving non-linear system an assertion failed during initialization.</p>
stdout | warning | <p>The non-linear solver tries to solve the problem that could take some time.</p>
stdout | warning | <p>It could help to provide better start-values for the iteration variables.</p>
stdout | warning | <p>For more information simulate with -lv LOG_NLS</p>
LOG_NLS | warning | <p>nonlinear system 192 fails: at t=0</p>
LOG_NLS | warning | <p>proper start-values for some of the following iteration variables might help</p>
LOG_NLS | warning | <p>[1] Real tank.s[2](start=2, nominal=1)</p>
LOG_NLS | warning | <p>[2] Real tank.s[1](start=2, nominal=1)</p>
LOG_NLS | warning | <p>[3] Real tank.ports[2].p(start=100000, nominal=1e+006)</p>
LOG_NLS | warning | <p>[4] Real radiator.statesFM[1].p(start=5e+006, nominal=1e+006)</p>
LOG_NLS | warning | <p>[5] Real radiator.statesFM[3].p(start=5e+006, nominal=1e+006)</p>
LOG_NLS | warning | <p>[6] Real radiator.mediums[1].p(start=110000, nominal=100000)</p>
LOG_NLS | warning | <p>[7] Real radiator.statesFM[2].d(start=150, nominal=500)</p>
LOG_NLS | warning | <p>[8] Real pipe.mediums[2].p(start=130000, nominal=100000)</p>
LOG_NLS | warning | <p>[9] Real pipe.mediums[2].h(start=335014, nominal=500000)</p>
LOG_NLS | warning | <p>[10] Real pipe.mediums[1].h(start=335014, nominal=500000)</p>
LOG_NLS | warning | <p>[11] Real pipe.mediums[1].p(start=130000, nominal=100000)</p>
LOG_NLS | warning | <p>[12] Real heater.mediums[1].p(start=130000, nominal=100000)</p>
LOG_NLS | warning | <p>[13] Real heater.mediums[1].h(start=335014, nominal=500000)</p>
LOG_NLS | warning | <p>[14] Real pump.medium.p(start=130000, nominal=100000)</p>
LOG_NLS | warning | <p>[15] Real radiator.m_flows[2](start=0, nominal=1)</p>
LOG_NLS | warning | <p>[16] Real radiator.mediums[1].h(start=167632, nominal=500000)</p>
LOG_NLS | warning | <p>[17] Real m_flow(start=0.01, nominal=1)</p>
LOG_NLS | warning | <p>[18] Real tank.medium.h(start=84013.1, nominal=500000)</p>
LOG_NLS | warning | <p>[19] Real pump.medium.h(start=167650, nominal=500000)</p>
LOG_NLS | warning | <p>[20] Real pump.N(start=1500, nominal=1)</p>
assert | debug | <p>Solving non-linear system 192 failed at time=0.<br>
For more information please use -lv LOG_NLS.</p>
assert | info | <p>simulation terminated by an assertion at initialization</p>
stdout | error | <p>Simulation process failed. Exited with code -1.</p>

beutlich, 2017-10-16:
@casella Is this somehow connected to #2060?

bilderbuchi, 2018-01-17:
I think `pump.delta_head_init >= 0.0` could be due to an error in `PartialPump`:
[Here](https://github.com/modelica/Modelica/blob/e60913e2120b35c5f8820099f9a77e846e91ebf6/Modelica/Fluid/Machines.mo#L316) we have the definition:
```
final parameter SI.Height delta_head_init = flowCharacteristic(V_flow_single_init)-flowCharacteristic(0)
```
However, at 0 volume flow, the pump head as determined by `flowCharacteristic` is at maximum, and decreases monotonically (which is checked by an [assertion](https://github.com/modelica/Modelica/blob/e60913e2120b35c5f8820099f9a77e846e91ebf6/Modelica/Fluid/Machines.mo#L626)). Therefore, for a positive initial volume flow, the head will be smaller, and therefore `flowCharacteristic(V_flow_single_init)-flowCharacteristic(0)` will be negative.

Does that sound correct?

bilderbuchi, 2018-07-10:
@casella is my above analysis/interpretation correct?

casella, 2018-08-18:
I guess the issue is just delta_head_init being of type SI.Height, which has  min = 0, a restriction that wasn't actually meant to be there. Based on the discussion in #2639, the problem should be solved by just changing SI.Height to SI.Length, I'll take care of that ASAP.

bilderbuchi, 2018-08-19:
I'm not so sure, as for a positive initial volume flow you get a negative initial delta head, which sounds like a sign error to me. (Where) is my previous analysis wrong?

bilderbuchi, 2018-10-18:
cross-referencing the equivalent OM issue: https://trac.openmodelica.org/OpenModelica/ticket/2958

beutlich, 2019-03-19:
@bilderbuchi Can you please check if #2834 fixes this issue? Thank you.

bilderbuchi, 2019-03-20:
Actually, I have never gotten confirmation either way of my [hunch/analysis](https://github.com/modelica/ModelicaStandardLibrary/issues/2370#issuecomment-358218010) that this is actually caused by a sign error in the delta_head_init calculation. If that is correct, the proposed change in #2834 from Height to Length will just mask/hide the error by not warning about the incorrect sign.
 
I can't check further because in current OMEdit, the above example is [structurally singular](https://libraries.openmodelica.org/branches/master/Modelica_trunk/files/Modelica_trunk_Modelica.Fluid.Examples.HeatingSystem.err). I have v1.13.0-dev.beta1 installed where this is also the case, but I could not find out when the example started failing.

beutlich, 2019-03-20:
Thanks for the update.

Seeing, that https://trac.openmodelica.org/OpenModelica/ticket/2958 still is unresolved, can someone from the OMC developers give a clue please if this is a tool issue or a library issue? @adeas31 @adrpo @casella @perost @sjoelund Thank you.

bilderbuchi, 2019-03-20:
That issue was reported by me, and after some time I concluded that it looks like an MSL issue, which is why I looked for the issue in the MSL tracker and found the present one.

HansOlsson, 2019-03-20:
> Actually, I have never gotten confirmation either way of my hunch/analysis that this is actually caused by a sign error in the delta_head_init calculation. If that is correct, the proposed change in #2834 from Height to Length will just mask/hide the error by not warning about the incorrect sign.

I agree and that issue is tool-independent - it is just that Dymola as default does not generate assertions for such values.

But what I find even more suspicious in Dymola is that:
pump.delta_head_init is -2.04m, pump.head_op is 2.04m and pump.head is 1.93 to 1.94m.

That looks very suspiciously as if pump.delta_head_init has the wrong sign.

casella, 2019-03-25:
Ok folks, sorry for the delay in the analysis, here it is.

The parameter `delta_head_init` is used to parameterize the simplified head-flow curve in the homotopy operator [here](https://github.com/modelica/ModelicaStandardLibrary/blob/8f65f621a388379e9de663fc15990bfc9ee9b540/Modelica/Fluid/Machines.mo#L437).

For stable operating points, this value is actually expected to be negative, so the model is correct, we just need to change the type `SI.Height` into `SI.Position` (which sounds better than  `SI.Lenght` in this case, and doesn't have the `min = 0` attribute as well)  everywhere in the pump model.

@beutlich, I can make a pull request about this both for master and 3.2.3 maintenance?

beutlich, 2019-03-25:
That is one of the changes in #2834 (and I asked @bilderbuchi to verify it in OMC).

HansOlsson, 2019-03-25:
That doesn't answer one implicit question: are pump.head_op and pump.head and pump.delta_head_init expected to have similar magnitude?

If so it seems odd that one of them has negative sign; even if the formulas are correct.
It may follow from conventions - but then it should (IMO) be explained in the model; and if it doesn't follow from usual conventions it would be better if they all had the same sign.

casella, 2019-03-27:
> That doesn't answer one implicit question: are pump.head_op and pump.head and pump.delta_head_init expected to have similar magnitude?

Yes.

> If so it seems odd that one of them has negative sign; 

Why, one is a the delta of the function (f(x) - f(0)), which happens to be negative if the function is monotonically decreasing. In fact, I could update the model to compute the delta on a smaller interval, since the idea is to approximate the derivative by a finite difference, which I currently do by taking (f(x) - f(0))/x. This won't affect the final result, only the homotopy transformation. In fact, that's how it's currently done in ThermoPower. At that point, the numerical value will be completely different and there won't be any doubt about it

> It may follow from conventions - but then it should (IMO) be explained in the model; and if it doesn't follow from usual conventions it would be better if they all had the same sign.

I could also improve the comments, explaining what it means.

I'll do it ASAP

casella, 2019-03-27:
Done, see #2834

beutlich, 2019-04-04:
Should be resolved after merging #2834.

closed 2019-04-04 by beutlich