modelica/ModelicaStandardLibrary #2371: Missing inheritance in Mechanics.Rotational.Components.Inertia

ceraolo, 2017-10-16:
According to the inheritance structure of MSL, it seems to me that it is advisable to substitute in the Inertia model the two rows:
```
  Rotational.Interfaces.Flange_a flange_a "Left flange of shaft";
  Rotational.Interfaces.Flange_b flange_b "Right flange of shaft";
```
with: 
`  extends Rotational.Interfaces.PartialTwoFlanges;
`

beutlich, 2017-10-16:
Resolved by b50fc360c3113b42bd872222149c697678e8fa1d in master.

The other question is, which flange description is the better one: "Left/Right flange of shaft" or "Flange of left/right shaft"?

ceraolo, 2017-10-16:
Since we have two shafts I vote for
Flange of left/right shaft"?

the other version was (only partially) acceptable for inertia, for which the two flanges cannot rotate from each other, but would be wrong for instance for disc.

beutlich, 2017-10-16:
Thanks. No need to change.

closed 2017-10-16 by beutlich