modelica/ModelicaStandardLibrary #2376: Modelica.Electrical.Spice3.Internal.Functions.energyGapDepTemp returns eV or V?

HansOlsson, 2017-10-24:
Modelica.Electrical.Spice3.Internal.Functions.energyGapDepTemp has an output with SI.Voltage - and computes it based on Spice3.Internal.MaterialParameters.EnergyGapSi - which is in GapEnergy which has unit "eV" (and means energy).

It doesn't seem right to mix the two. 

If it should be an actual energy then many other functions need to be changed as well; related to #2375.

kristinmajetta, 2017-12-14:
changed unit from voltage to GapEnergy in Modelica.Electrical.Spice3.Internal.Functions.energyGapDepTemp
SHA-1: c72fe9481999a4e29f1dad8d7a5bb9b707963aa1

beutlich, 2018-01-02:
@kristinmajetta Can this be closed then or why is this issue left open?

HansOlsson, 2018-06-11:
This change needs to be propagated further.

After this change Dymola generates unit-warnings for Modelica.Electrical.Spice3.Examples.Inverter, Modelica.Electrical.Spice3.Examples.InvertersApartRecord, Modelica.Electrical.Spice3.Examples.InvertersExtendedModel, Modelica.Electrical.Spice3.Examples.Spice3BenchmarkFourBitBinaryAdder due to Modelica.Electrical.Spice3.Internal.Functions.junctionParamDepTempSPICE3 and Modelica.Electrical.Spice3.Internal.Functions.saturationCurDepTempSPICE3MOSFET:

The change in junctionParamDepTempSPICE3 just seems to be to change phibtnom and phibtemp to eV as well. 

However, saturationCurDepTempSPICE3MOSFET seems messier, clearly energygaptemp and energygaptnom should have type EnergyBut, and since it uses `exp( energygaptnom / vtnom - energygaptemp / vt)` it seems that also vt and vtnom should be in GapEnergy, but they are given as:
```
 vt := Modelica.Electrical.Spice3.Internal.SpiceConstants.CONSTKoverQ*temp;
 vtnom := Modelica.Electrical.Spice3.Internal.SpiceConstants.CONSTKoverQ*tnom;
```
I don't know if both sides are gap-energies in this case.

kristinmajetta, 2018-06-12:
solved unit issues in functions Modelica.Electrical.Spice3.Internal.Function.junctionPotDepTemp, Modelica.Electrical.Spice3.Internal.Function.junctionParamDepTempSPICE3 and Modelica.Electrical.Spice3.Internal.Function.saturationCurDepTempSPICE3MOSFET
in SHA-1: 783b3c1dedd5708499162f30dce3a9c048b65655

PR #2619 (2018-06-15): Resolve Spice3 unit issues according to #2376 in functions
Modelica.Electrical.Spice3.Internal.Function.junctionPotDepTemp
Modelica.Electrical.Spice3.Internal.Function.junctionParamDepTempSPICE3
Modelica.Electrical.Spice3.Internal.Function.saturationCurDepTempSPICE3MOSFET

close #2376
closed 2018-06-15 by kristinmajetta