modelica/ModelicaStandardLibrary #2377: What is the unit of R?

HansOlsson, 2017-10-24:
While correcting #2375 I found that Modelica.Media.Interfaces.PartialMedium contains SpecificHeatCapacity R (with unit J/(kg.K) ) and Modelica.Constants.R is 8.3144598 J/(mol.K).

On its own this is confusing - and likely an error. And at least an actual error is caused by this in:
Modelica.Media.Incompressible.TableBased.BaseProperties where R=Modelica.Constants.R

thorade, 2017-10-24:
Modelica.Constants.R is the universal gas constant, 
the other one is the special gas constant, multiplied by the molar mass of the substance in question.

HansOlsson, 2017-10-24:
> Modelica.Constants.R is the universal gas constant, the other one is the special gas constant, multiplied by the molar mass of the substance in question.

To me this indicates that "the other one" should have another name - since using R for both of them seems error-prone.

thorade, 2017-10-24:
Please note that I had to correct my comment above (divide/multiply):
R_s = R/MM
[J/kg.K] = [J/mol.K] / [kg/mol]

HansOlsson, 2017-10-24:
> Please note that I had to correct my comment above (divide/multiply): 
> R_s = R/MM

Noted, and another important change is that you introduced R_s for the substance(?) gas constant; avoiding the confusion of two R that are different. Is that normal - or is another term used to disambiguate it?

thorade, 2017-10-24:
The correct term is specific gas constant, it seems:
https://en.wikipedia.org/wiki/Gas_constant#Specific_gas_constant
I think I have seen R_s as variable name, but usually people stick with mass based units or with molar units for R and density within one publication, so that they do not have to disambiguate.

thorade, 2017-10-24:
And other people do it the other way round and refer to the universal or molar gas constant as R_m.

HansOlsson, 2017-10-24:
> And other people do it the other way round and refer to the universal or molar gas constant as R_m.

Since NIST https://physics.nist.gov/cgi-bin/cuu/Value?r use the term R - and even I remember pV=nRT that seems like a less good solution.

However, we can wait with the renaming as long as Modelica.Media.Incompressible.TableBased.BaseProperties is corrected to not use R=R; - for R with different units.

thorade, 2017-10-24:
Some people (e.g. engineers) just prefer mass based units,
other (e.g. chemists) might prefer molar units.

The ideal gas law can be written using mass based or molar units units:
https://en.wikipedia.org/wiki/Ideal_gas_law#Molar_form
Especially, the last sentence of that paragraph!

hubertus65, 2018-01-21:
I think this has to be split into 2 issues: Fixing the naming affects all Media models and is probably not really backwards compatible. Fixing TableBased properties is a bug and should be fixed now. Will do when I get to it.

thorade, 2019-09-06:
Note: The longer term ticket for naming is #2438

thorade, 2020-01-31:
Just a note: https://github.com/usnistgov/REFPROP-issues/issues/239

casella, 2020-01-31:
Interesting to hear that now R is fixed, as the Avogadro number and the Boltmann constants have also been fixed. I wasn't aware of that.