modelica/ModelicaStandardLibrary #2379: EddyCurrent using non-resistance resistance-function

HansOlsson, 2017-10-24:
The tests for #2375 detected another issue that required a bit extra to identify.
The models
Modelica.Mechanics.Rotational.Sources.EddyCurrentTorque
Modelica.Mechanics.Translational.Sources.EddyCurrentForce
use Modelica.Electrical.Machines.Thermal.convertResistance - for something that isn't a Resistance, but just a number with unit "1".

Thus it fails a unit-check. 

I don't know if the solution is to add a variant of convertResistance (with better name) - or remove unit from the existing convertResistance.

MartinOtter, 2017-10-24:
These are models from Anton Haumer

AHaumer, 2018-01-03:
I'll add a function linearTemperatureDependency which does the same as convertResistance but without physical units for input RRef and result R.

commit 5861158e (2018-01-03): resolves #2379
closed 2018-01-03 by AHaumer