modelica/ModelicaStandardLibrary #2385: Incorrect assert statement in IF97_Utilities.BaseIF97

greenwoodms06, 2017-11-02:
Assert statement checks if T >= 273.16 and then uses `triple.ptriple` (when it should use `triple.Ttriple`) 
- Action: `ptriple` was changed to `Ttriple` - line ~3020

HansOlsson, 2017-11-02:
Good catch, but wrong solution I think.

I believe you refer to Modelica.Media.Water.IF97_Utilities.BaseIF97.Basic.psat with:
```
assert(T >= 273.16, "IF97 medium function psat: input temperature (= "
     + String(triple.ptriple) + " K).\n" +
    "lower than the triple point temperature 273.16 K");
```

That should be `T` - shouldn't it? But possibly change 273.16 to `triple.Ttriple`.

greenwoodms06, 2017-11-02:
@HansOlsson Yes that is the line. Based on the current phrasing yes I think you are correct. I updated the original post to reflect your comment.

PR #2497 (2018-04-11): Fixes for issues #2385, #2490 and #2499
Fixes for issues #2490 (a one letter typo) and #2385 (a one letter typo that is significant, a wrong assert).

Close #2490. Close #2385. Close #2499.
closed 2018-04-11 by beutlich