modelica/ModelicaStandardLibrary #2388: additonal machines example

AHaumer, 2017-11-08:
Add an additonal example for a synchronous machine with damper cage, starting direct on line, reaching synchronous speed. Loaded with a (rather small) load torque, it keeps synchronous speed.

dietmarw, 2017-11-09:
I looked at the example but what really would be nice is to not have a reluctance machine running up but a standard synchronous motor with damper cage where you then switch on the field current once near enough synchronous speed. This is something that would be nice to demonstrate to students.

AHaumer, 2017-11-12:
Good point Dietmar! I'll have a look.

AHaumer, 2017-11-12:
Feel free to develop new examples and provide them ...

dietmarw, 2017-11-14:
@AHaumer I looked at the EE example but it is not really showing the expected behaviour. That machine simply runs asychronously and does not proper ly synchronise. The idea should be that it starts up asychronously and then "locks" to the synchronous field by activating the excitation. So the speed plot should clearly show that the machine runs at synchronous speed before you load it. Currently it does not do that.

dietmarw, 2017-11-14:
@AHaumer I played around a bit and I suggest to change the following parts (I don't want to do it directly in case you disagree):
In `torqueStep`:
```diff
- startTime=1, 
- stepTorque=100) 
+ startTime=2, 
+ stepTorque=50) 
```
This means the synchronous machine has time to synchronise and the load will not over load it so that the displacement angle is below 30 degrees.

christiankral, 2017-12-10:
When investigating example ```Modelica.Electrical.Machines.Examples.SynchronousInductionMachines.SMEE_DOL```, I noticed, that the mains switch closes at time = 0.1 s. So when I watch the rotor displacement angle ```rotorDisplacementAngle.rotorDisplacementAngle``` it shows a severe noise before the switch closes, i.e., for time < 0.1 s. This noise is not physical. 

The rotor displacement angle is calculated as the difference between a) the angle of the stator terminal voltage space phasor and b) the mechanical angle, multiplied by the number of pole pairs. So when the mains switch is open, the stator terminal voltages show mainly noise, caused by the interaction of the high switch resistance and the machine models, supplied by the mains voltages. Therefore, the noise of the stator terminal voltages for time < 0.1 s causes the rotor displacement angle also to show non-physical noise. 

In an ideal case with infinitely high switch resistances, the stator terminal voltages were exactly equal to zero. In this ideal case the rotor displacement angle were zero, too. 

So on my opinion there are two options to handle this noise issue of this model:
1. Explain to the users, that the rotor displacement angle has no physical meaning for time < 0.1 s  
2. Close the switch at time = 0 -- and explain to the user why the switch close as time = 0

Particularly, this issue became obvious, when I started to add a similar example to the FundamentalWave machine examples. Then, the two examples showed different rotor displacement angles for time < 0.1 s, which is NOT nice on my opinion. I will therefore implement option 2 (in the existing and the new model ```Modelica.Electrical.Machines.Examples.SynchronousInductionMachines.SMEE_DOL```).

christiankral, 2017-12-10:
Implemented option 2, added new model in ```FundamentalWave``` package.

AHaumer, 2017-12-10:
Fine for me, many thanks Christian!

AHaumer, 2017-12-11:
I'll try to adapt the rotorDisplacement model to suppress the small nearly random voltage that goes over the open switch to avoid similar problems in other experiments.

commit 3f955497 (2017-11-08): resolves #2388
commit 382837b6 (2017-11-12): fixes #2388 (D.Winkler)
commit c08437b7 (2017-11-29): fixes #2388
commit 825c14bd (2017-12-11): resolves #2388
closed 2017-12-11 by AHaumer