modelica/ModelicaStandardLibrary #2394: Error in documentation for InlineAfterIndexReduction

GarronFish, 2017-11-16:
On page 233 in ModelicaSpec34.pdf

Documentation states:
[This annotation is for example used in Modelica.Mechanics.Rotational.Sources.Move to define that an input signal is the derivative of another input signal.]

However now LateInline=true is used since modifications in ticket #1523.

beutlich, 2017-11-16:
Looks like this should have been reported at https://trac.modelica.org/Modelica/.

GarronFish, 2017-11-16:
Which tickets should be reported at https://trac.modelica.org/Modelica/ and which should be report here?

beutlich, 2017-11-16:
Library tickets go here, Specification tickets remain on Trac.

HansOlsson, 2018-01-15:
This is now migrated to: https://trac.modelica.org/Modelica/ticket/2222
@GarronFish I did not CC you on the Modelica-ticket; if you want that it would be straightforward.

closed 2018-01-15 by HansOlsson