modelica/ModelicaStandardLibrary #2396: Access violation in ModelicaInternalC

niklwors, 2017-11-20:
With Viusal Studio 2013 we get an access violation in ModelicaInternalC in function ModelicaStreams_openFileForReading.
The access vioaltion happens for the call MUTEX_LOCK() in line 4 of this function.
The problem is that initializeCS() is not called before l MUTEX_LOCK()  is called. The problem occurs since the last changes for gconstructor.h in case 
#elif defined(_MSC_VER) && (_MSC_VER >= 1500) /* Visual Studio 2008 and later has _pragma */
With the old code version for this case it is working fine.

beutlich, 2017-11-20:
> The problem occurs since the last changes for gconstructor.h

Which *last* changes exactly? Please point to the commit you refer to?

Did you see #2073 and #2074?

niklwors, 2017-11-20:
The problem happens since commit e2e1dbc7b11b0604afd927ccb451e5064a937c4d
I looked at your mentioned issues but I think this is not related to the problem

beutlich, 2017-11-20:
I tried to reproduce with VS2015 but failed. Please find my VS solution [T2396.zip](https://github.com/modelica/Modelica/files/1488894/T2396.zip) attached. You might want to try it on your own.

niklwors, 2017-11-21:
The problem is that we compile the ModelicaExternalC source files to a static library and the linker only prints this as a warning instead of an error:
ModelicaInternal.c.obj : warning LNK4006: __arrayinitializeCS already defined in ModelicaRandom.c.obj; second definition ignored
ModelicaInternal.c.obj : warning LNK4006: _initializeCS_wrapper already defined in ModelicaRandom.c.obj; second definition ignored
ModelicaInternal.c.obj : warning LNK4006: __arraydeleteCS already defined in ModelicaRandom.c.obj; second definition ignored
ModelicaInternal.c.obj : warning LNK4006: _deleteCS_constructor already defined in ModelicaRandom.c.obj; second definition ignored

Is this because of wrong linker flags or is this a problem in the ModelicaExternal C source code?

beutlich, 2017-11-21:
I fixed the linker warnings (for LIB) / erros (for EXE) by d762a763a03429239b1aaa89f4e4531c92118af4 in master. Does it already help for you scenario?

niklwors, 2017-11-22:
Unfortunately, the problem still exists.
 I noticed that I also get this warnings
ModelicaInternal.c.obj : warning LNK4006: _opendir already defined in win32_dirent.c.obj; second definition ignored
ModelicaInternal.c.obj : warning LNK4006: _readdir already defined in win32_dirent.c.obj; second definition ignored
ModelicaInternal.c.obj : warning LNK4006: _closedir already defined in win32_dirent.c.obj; second definition ignored
ModelicaInternal.c.obj : warning LNK4006: _rewinddir already defined in win32_dirent.c.obj; second definition ignored
ModelicaInternal.c.obj : warning LNK4006: _telldir already defined in win32_dirent.c.obj; second definition ignored
ModelicaInternal.c.obj : warning LNK4006: _seekdir already defined in win32_dirent.c.obj; second definition ignored
But I think this is not related to the problem

beutlich, 2017-11-22:
This is indeed a different problem. Check out how I solved it in my example VS solution above. Since win32_dirent.c is included by ModelicaInternal.c it must not be compiled as own module.

niklwors, 2017-11-22:
> Since win32_dirent.c is included by ModelicaInternal.c it must not be compiled as own module.
Thank you, the problem was also that we added win32_dirent.c to the build for ModelicaExternalC. I removed it and now it is working.

closed 2017-11-22 by beutlich