modelica/ModelicaStandardLibrary #2397: A better Multiphase power sensor

ceraolo, 2017-11-25:
The power sensor in the Electrical.Analog.MultiPhase sensor uses many variables and equations and requires adding a star and its ground.
Indeed because of Aron's theorem, power measure can be made in a simpler way, which uses much less variables and allows cleaner diagrams.
I enclose a package in which I propose the new AronSensor and two example models.
These two example models do exactly the same thing, but AronPower is simpler. Moreover, StdPower has 210 equations, while AronPow has 121.

[Pow3Sensors.zip](https://github.com/modelica/Modelica/files/1502944/Pow3Sensors.zip)

AHaumer, 2017-11-26:
To be correct;
This isn't the Aron connection we're used from threephase systems which uses 2 power meters instead of 3. https://de.wikipedia.org/wiki/Aronschaltung#/media/File:Leistung-AronschaltungWB.svg
The upper image shows active power measurement, the lower image shows reactive power measurement in a threephase system.
Your "AronSensor" calculates power in all phases and relies on the fact that either source or load are grounded. Using a power sensor in a real system means you have to connect not only the current branch but also the correct voltage branch. 
Regarding the count of equations, the difference can be found only looking at the alias variables i.e. trivial equations.
In your example, you can delete star1 and ground1 and connect powersensor.nv to either star.plug_p or star2.plug_p. So you just save 2 connections (to powerSensor.pv and powerSensor.nv).

ceraolo, 2017-11-27:
Thank you for considering this post. AHaumer.
Indeed I agree that the name "AronSensor" at least needs some explanation, and could be even changed, if that sensor (or a modified version of it) is adopted.

I did not comment on the name in my original post, because I first wanted to see if it stimulated discussion or not.
In the beginning I choose the name "AronSensor" because the icon of the proposed model on a diagram is like "as a set of Aron wattmeters": along the line, as  in the pictures from wikipedia you linked.
But that Aron connection gives correct results only for "balanced" three-wire three-phase systems (null sum of currents). So I resorted to the generalised formulation of Aron's theorem, that states that if a circuit is composed by two parts, connected through m+1 wires, the total power absorbed by one part is given by the sum of products of m voltages and  m currents, the voltages being taken with reference to the (m+1)-th wire (for which current is not measured).
So my "AronSensor" is an Aron Sensor in two ways:
1) it appears as a component though  the line, indicating a set of individual wattmeters (as two Aron sensors when the system is balanced) 
2) still makes usage of the Aron theorem (in its generalised form), also for imbalanced m-phase systems, but hides the inner usage of potentials instead of differences of potentials.

Maybe the name can be changed and this explanation be moved in the model information.

**Regarding conneting powersensor.nv to star.plug_p.**  
It reduces the number of equations and connections, but gives rise to a drawing that is farther from the actual situation. The power sensor might be for instance in the middle of a several-km line, and in that case the measure would be taken from local measures, while your connections would imply a long measuring line connecting plugs.

**Regarding equation count.** 
Maybe both total # of equations and # of non-trivial equations impact on memory and cpu-time, although the latter are much more important

We can count the non-trivial equations as measured by OpenModelica. (StdPow2 is with connection of powersensor.nv to star.plug_p).  

Model                        total         non-trivial 
AronPow                      121                24
StdPower                      71                 34
StePower2                    64                 36

So, to conclude my proposal:
- requires minimal effort to put a sensor in the diagram
- has the aspect of watt-meters along-the line, just as Aron sensors
- reduces significantly both equations and nontrivial equations
- works with balanced and imbalanbed systems

The name "AronSensor" might be tricky and require to be changed, although its derivation from generalised Anon's theorem can be mentioned in the block information.

Therefore I still see good reasons for considering its inclusion in MSL.

ceraolo, 2017-11-27:
Sorry, but I must add something more (then I will shut up!)

After some more thoughts I must admit that the model I called "AronSensor" is not reliable enough for general users and imbalanced systems: it does not work well with imbalanced systems and return path with non-zero resistance.
I withdraw it as a proposal for MSL.

But a small modification allows to create a new one that in Pow3SensorsNew.mo is called AronSens.
[Pow3SensorsNew.zip](https://github.com/modelica/Modelica/files/1507536/Pow3SensorsNew.zip)
It is still faster to use than the MSL Power sensor, creates less equations and less nontrivial equations and leaves the diagram cleaner. It is easer to understand.

It has not the disadvantage of working badly with imbalanced systems.

It seems to me that this one has only advantages over the standard PowerSensor of MultiPhase library.

AHaumer, 2017-11-28:
I don't like it from the didactic point of view: in reality you have to provide current branch and voltage branch. Measurement in reality means wither using the Aron 2 wattmeter method, or using the neutral point if available or using an artificial starpoint. The latter is common practice when measuring at power electronics.
I'd like to be able to determine which voltage to be used. 
You're introducing a source fpr numerical rounding errors: calculating power with the possibly much higher potential (think of a potential shift).
Why bother about a arther small amount of equations? There are other things that influence simualtion speed much more (stiff system, proper index reduction, event iteration, tearing of really large systems, ...)
If you would like to have such a simple-to-use device (maybe with pre-connected paths), why not create a power analyzer (e.g. https://valhallascientific.com/wattmeters-power-analyzers/2800-series/ or http://www.infratek-ag.com/infratek---108a.html) which measures voltages and currents, determines frequency, calculates active, reactive and apparent power, power factor, root mean square and even hamronics.

ceraolo, 2017-12-04:
Ok.
I see there is not support to the idea, so better to close the ticket now
Thanks to all that have contributed.

dietmarw, 2017-12-04:
I would not close this right away. The two-power meter method is still a relevant method and also taught in Power Systems. I think it would still proof as a valuable addition but not replacement. So please leave this open for the time being.

AHaumer, 2017-12-05:
I agree with Dietmar. A two power meter instrument (Aron) for threephase systems is a good addition, I support this idea We could even provide an instrument for reactive power.

AHaumer, 2018-01-04:
Wait we forgot the quaistationary version ...

dietmarw, 2018-01-04:
And an example and test version. ;-)

AHaumer, 2018-01-04:
There are examples both for transient and quasistationary (TestSensors)

closed 2018-01-04 by AHaumer

max-privato, 2019-09-30:
I've seen that in Modelica 3.2.3 an Aron sensor has appeared. 
Although the implementation is different, the derivation is evidently from my proposal.
I would have liked to be mentioned in the info of that component.

beutlich, 2019-10-05:
@max-privato Feel free to provide the appropriate pull request. Thank you.

max-privato, 2019-10-06:
_Feel free to provide the appropriate pull request. Thank you._
I will do. 
I will also propose to correct the comment technically: It currently says that 
_This device works only for sinusoidal voltages and currents in threephase systems without neutral._
in reality this is valid also for non-sinusoidal quantities.

commit aee4c021 (2018-01-04): resolves #2397
commit 685cc949 (2018-01-04): resolves #2397 (quasistationary)