modelica/ModelicaStandardLibrary #2404: MSL - text file format of external input data sets

ToSchwan, 2017-12-11:
Dear community,

the Modelica Standard Library (MSL) currently provides three ways to integrate external data sets into simulation. Those are mainly the three table types in Modelica.Blocks.Tables directory. These models enable the input of external *.txt-files and integrated tables as an additional data set.
I' ve recently recognized that these *.txt-files' format must be UTF-8 without BOM (Byte-Order-Mark) by default. However, newly developed text-file generators (e.g. in Python) usually create text-files in UTF-8 (with BOM). Normal text editors are not able to show or adapt these files' format afterwards (I know only Notepad++). To find this issue took me very long. Maybe, the topic has already been recognized by other users and could be fixed within upcoming library updates.

Thanks for your help.
Torsten

beutlich, 2017-12-11:
@ToSchwan Thanks for the suggestion.

@HansOlsson The requirement that text file need to start with `#1` came from the Dymola implementation of the MSL v3.2 tables. Do you agree that we should extend this format from plain ASCII text to ASCII and UTF-8 (with or w/o BOM) encoded text?
* Comments are ignored anyway
* Table names still must not use UTF-8 two byte characters

thorade, 2017-12-11:
As you mention Notepad++ explicitly: It is able to open both UTF-8 with or without BOM, as well as most other encoding, and it is able to convert between the encodings. See screenshot below:

[image]

ToSchwan, 2017-12-11:
Hi,
thank for that quick answer. I already know that functionality in Notepad++. And it's feasible for a reduced number of generated text files to convert them manually with Notepad++. However, this approach crashes in case of hundreds of generated files. This way, an adaption in the MSL would be nice to have.

Thank you.

thorade, 2017-12-11:
And another note: To my knowledge, Python would usually write utf-8 without BOM.
You can even use it to convert from utf-8 with BOM to utf-8 without BOM, see example below.
That example could easily be extended to convert all txt files in a directory using `os.walk`.

```python
from io import open  # backported from Py3 to Py2
import os

cwd = os.getcwd()
finPath = os.path.join(cwd, "test.txt")
foutPath = os.path.join(cwd, "fout.txt")

with open(finPath, mode="r", encoding="utf-8-sig") as fin, open(foutPath, mode="w", encoding="utf-8") as fout:
    for line in fin:
        line.rstrip()
        fout.write(line)
```
utf-8-sig will skip the BOM (if present)
https://docs.python.org/3/howto/unicode.html#reading-and-writing-unicode-data

beutlich, 2017-12-12:
The suggested enhancement is to simply also allow `BOM#1` as valid first bytes for a text file. I believe this is an acceptable anhenacement.

beutlich, 2018-01-09:
@HansOlsson Friendly reminder.

HansOlsson, 2018-01-12:
> @HansOlsson The requirement that text files need to start with #1 came from the Dymola implementation of the MSL v3.2 tables. 

I checked, and it goes back at least 19 years, and the '1' is a version number which hasn't changed.

> Do you agree that we should extend this format from plain ASCII text to ASCII and UTF-8 (with or w/o BOM) encoded text?

Yes, ignoring the BOM seems just seems right.

PR #2428 (2018-01-17): Ignore UTF-8 BOM of text file for table blocks
Close #2404.
closed 2018-01-17 by beutlich