modelica/ModelicaStandardLibrary #2406: MassFractionsTwoPort — code bug

justnielsen, 2017-12-20:
In `Modelica.Fluid.Sensors.MassFractionTwoPort` there is a `for` loop in the `initial algorithm` section iterating from `1:Medium.nC`. This should be changed to `1:Medium.nXi`.

beutlich, 2017-12-20:
This line: https://github.com/modelica/Modelica/blob/23ce318d3e30ac2972ee140825c482dc874dd172/Modelica/Fluid/Sensors.mo#L462

sjoelund, 2017-12-20:
Could simply be an initial equation without algorithm or anything fancy:
```mo
assert(max(name == substanceName for name in Medium.substanceNames), "Mass fraction '" + substanceName + "' is not present in medium '"
           + Medium.mediumName + "'.\n"
           + "Check sensor parameter and medium model.");
```

beutlich, 2017-12-20:
The `max` seems odd. Is there a `break` statement missing in all these loops (also MassFractions, TraceSubstances and TraceSubstancesTwoPort)?

sjoelund, 2017-12-20:
`max` may seem weird, but it's the Modelica way of checking if some element of an array is true. It also avoids temporary variables which become part of result files (if protected variables are included).

wischhusen, 2018-01-04:
We need "ind" below in the equation part to access the right substance. I think I will stick to the existing initial algorithm. But nice idea.

beutlich, 2018-01-09:
@wischhusen Will you file a PR from your branch or simply merge it yourself?

PR #2420 (2018-01-09): Pull request for branch Issue2406
Close #2406.
closed 2018-01-09 by wischhusen