modelica/ModelicaStandardLibrary #2407: CombiTable2D tests with unknown variables in reference results

henrikt-ma, 2017-12-20:
The following tests in RegressionTesting/ReferenceResults/MSL/trunk/ModelicaTest/Tables/CombiTable2D have unknown `surface.y[…]` variables in the reference results:
- Test15
- Test16
- Test17

A suppose these come from a vendor specific implementation of ModelicaServices, and I guess the lists of comparison variables for these tests should simply be empty.

sjoelund, 2017-12-20:
Only variables that are part of `Modelica.Utilities.Internal.PartialModelicaServices.Animation.PartialSurface` should be present, right?

beutlich, 2017-12-20:
Yes, but these variables are of no use for the regression here.

sjoelund, 2017-12-20:
Those are the only variables though...

GallLeo, 2018-08-27:
I now emptied the list of comparison signals. As there are no public variables, I don't think result comparison does make a lot of sense for these test cases.

Ticket can be closed after upload of new reference results.

closed 2018-11-22 by beutlich