modelica/ModelicaStandardLibrary #2408: Dymola Error Funtcion with Complex

Magdaanne, 2017-12-29:
I get the following Error:
_Assignment to parameter or constant: A_

also for the other protected Complex B; C; .... in the function:

  _import Complex;
  input Real omega;
  input Real RaM;
  input Real RbM;
  input Real Ro1M;
  input Real Ro2M;
  input Real CaM;
  input Real CbM;
  input Real CcM;
  output Complex A_o;
  output Complex B_o;
  output Complex C_o;
  output Complex D_o;
protected 
 parameter Complex A;
 parameter Complex A1;
 parameter Complex B;
 parameter Complex B1;
 parameter Complex C;
 parameter Complex C1;
 parameter Complex D;
 parameter Complex D1;
 parameter Integer i;
algorithm 
  for i in 1:7 loop
    A := Complex(1, 0);
    D := Complex(1, 0);
    if i == 1 then
      B := -RaM * Complex(1, 0);
      C := Complex(0, 0);
    elseif i == 2 then
      B := Complex(0, 0);
      C := Complex(0, 1) * (-CaM * omega);
    elseif i == 3 then
      B := (-Ro1M) * Complex(1, 0);
      C := Complex(0, 0);
    elseif i == 4 then
      B := Complex(0, 0);
      C := Complex(0, 1) * (-CcM * omega);
    elseif i == 5 then
      B := (-Ro2M) * Complex(1, 0);
      C := Complex(0, 0);
    elseif i == 6 then
      B := Complex(0, 0);
      C := Complex(0, 1) * (-CbM * omega);
    elseif i == 7 then
      B := (-RbM) * Complex(1, 0);
      C := Complex(0, 0);
    end if;
    if i == 1 then
      A_o := A;
      B_o := B;
      C_o := C;
      D_o := D;
    else
      A1 := A * A_o + B * C_o;
      B1 := A * B_o + B * D_o;
      C1 := C * A_o + D * C_o;
      D1 := C * B_o + D * D_o;
      A_o := A1;
      B_o := B1;
      C_o := C1;
      D_o := D1;
    end if;
  end for;
end cal_mod;_

when I don't define A; B ... as a parameter I get Errors using the function..
The model was originally written for OpenModelica but I want to use it in Dymola and I can't find the difference in syntax

I'm grateful for every help!

beutlich, 2018-01-02:
This is not the correct spot when asking for help. Anyway, declaring the protected parameters as variables should do the trick.

closed 2018-01-02 by beutlich

Magdaanne, 2018-01-02:
Hey :) when I declare the protected parameters as variables, I get another Error in the model, where I'm using this function. It does run in OpenModelica, but not in Dymola. 
And I'm sorry this isn't the correct spot, where can I ask for help?

HansOlsson, 2018-01-02:
> And I'm sorry this isn't the correct spot, where can I ask for help?

I would say www.stackoverflow.com with tag `[modelica]` and try to include a complete example demonstrating the issue.