modelica/ModelicaStandardLibrary #2409: Copy-paste error breaks MSL 3.2.2 models

sjoelund, 2018-01-02:
db89e60206646c9a9d89a314c0fc1b012c903671 (#2301) introduced connections without introducing the `lessThreshold` component, which suggest a copy-paste or cherry-pick error:
```mo
connect(ToPolarVSR.y, lessThreshold.u);
connect(lessThreshold.y, wrapAngle.u);
```

The affected models are in an OpenModelica [report](https://libraries.openmodelica.org/branches/history/master/2017-12-31%2004:57:10..2018-01-01%2000:30:40.html); @christiankral please test these models when making a fix in MSL.

beutlich, 2018-01-02:
0d69a2c5d38ba6751292b8a43d4d98b4f4a86169 is the breaking commit.

christiankral, 2018-01-02:
It seems that I overlooked a non-working model which caused the trouble with breaking the MSL. Sorry for that. Everything shall be fixed by now. I double checked all changes of master and maint/3.2.2.

closed 2018-01-02 by christiankral