modelica/ModelicaStandardLibrary #2410: PowerConvert templates with current sensors

HansOlsson, 2018-01-02:
When looking for current use of current-sensors based on #2301 I noticed some models that looked odd:

Modelica.Electrical.PowerConverters.Examples.DCDC.ExampleTemplates.ChopperStepDown
Modelica.Electrical.PowerConverters.Examples.DCAC.ExampleTemplates.SinglePhaseTwoLevel
(possibly more of the same kind.)

They have a voltage sensor in series with a current sensor which is something you wouldn't normally expect, until you read the description and find that load is not included and then see examples of its use and realize that the load should be in parallel to the current sensor. 

One possible improvement would be to mark these as `partial` models (since they are only intended to be used as base-classes; in contrast to a similar model having load as a replaceable component - that can also be used as a sub-component with suitable redeclare), and another would be to have nodes indicating where to connect the load.

christiankral, 2018-01-02:
All templates but one were partial. It was only ```ThyristorCenterTap2Pulse``` which was missing the partial indication.

christiankral, 2018-01-02:
Now all template models are partial.

AHaumer, 2018-01-03:
I see no problem that in a partial (!) model a voltage sensor and a current sensor seem to be connected in series since the load still is missing. In the example which extends from the partial template the voltage sensor is connected in parallel and the current sensor in series to the load.

closed 2018-01-03 by AHaumer

HansOlsson, 2018-01-04:
> I see no problem that in a partial (!) model a voltage sensor and a current sensor seem to be connected in series since the load still is missing. In the example which extends from the partial template the voltage sensor is connected in parallel and the current sensor in series to the load.

I agree - if you can see that a load is missing there. 
I could figure that out, but it wasn't obvious at first sight.

Thus the suggestion was more to consider making it clearer that a load should be connected at a specific spot for people less familiar with the library (e.g. by adding nodes indicating connection points - or just documenting this). But I don't insist on this, since too much "hand-holding" for novices can also make the library seem trivial for power users.