modelica/ModelicaStandardLibrary #2417: PowerConverters

AHaumer, 2018-01-06:
I have some minor enhancements to PowerConverters, as well as an implementation of a step up (boost) converter. I'll create a pull request.

dietmarw, 2018-01-07:
@AHaumer You normally don't need to open up an additional issue for a PR since the PR itself serves as an issue with a discussion thread. So simply create a PR and describe what it does and its purpose which then can be discussed and approved by somebody else to be accepted. Just saves a bit of work ;-)

beutlich, 2018-01-09:
Can this be closed then?

closed 2018-01-09 by christiankral

beutlich, 2018-01-10:
Can you please edit the issue caption to make clear what the exact enhancement is? For example: Add step up chopper to PoverConverters. This makes it easier when collecting the relevant information for the release notes.