modelica/ModelicaStandardLibrary #2430: Obsolete adaptors in Mechanics.Rotational.Component

tobolar, 2018-01-17:
@AHaumer: In 49e9cdbf9cc40f2c7b11b68306b4bc7c0d4cefa9 there have been added two adapters into Rotational package: `Modelica.Mechanics.Rotational.Components.TorqueToAngleAdaptorNew` and `Modelica.Mechanics.Rotational.Components.AngleToTorqueAdaptorNew`. What about the original ones (`Modelica.Mechanics.Rotational.Components.TorqueToAngleAdaptor` and `Modelica.Mechanics.Rotational.Components.AngleToTorqueAdaptor`)? Should they be marked as obsolete?

beutlich, 2018-01-18:
Furthermore, the new names are questionable. For example, there also is Modelica.Electrical.Analog.Semiconductors.Diode and its improved implementation Diode2.

tobolar, 2018-02-05:
See #2440 for `Modelica.Mechanics.MultiBody.Examples.Loops.Utilities.GasForce`.

beutlich, 2018-05-03:
Idea: Set defaultComponentName of new classes as old name to not have the "New" in the instance name.

GallLeo, 2018-05-03:
I don't think the old ones are obsolete. The two solutions should be seen as alternatives. 
There are many ways to write adaptors for FMUs, and MSL can only give examples/hints.

The "old ones" by @MartinOtter have the huge benefit of being one flat class, which you can directly read and understand.
Already these "old ones" include many Modelica tricks like `Evaluate=true` combined with protected variables `*_internal` plus combining the derivatives of the inputs in the equation section:

Old (MSL 3.2.2):
``` modelica
equation 
  connect(w, w_internal);
  connect(a, a_internal);
  phi = flange.phi;
  if use_w then
    w_internal = der(phi);
  else
    w_internal = 0.0;
  end if;
  if use_a then
    a_internal = der(w_internal);
  else
    a_internal = 0.0;
  end if;
  flange.tau = tau;
```

The new ones from @AHaumer are implemented in a more general way, but have the drawback that connectors are named `p`, `pder` and `pder2` instead of `phi`, `w` and `a`.
But the main issue is, that the new ones use Functions.state2() which is quite complicated to read.

New (master):
``` modelica
equation 
  y = p;
  y1 = if use_pder then der(y) else 0;
  y2 = if (use_pder and use_pder2) then der(y1) else 0;
  connect(y1, pder);
  connect(y2, pder2);
  if use_fder then
    connect(fder, u1);
  else
    u1 = 0;
  end if;
  if (use_fder and use_fder2) then
    connect(fder2, u2);
  else
    u2 = 0;
  end if;
  if (use_fder and use_fder2) then
    u = Functions.state2({f, u1, u2}, time);
  elseif (use_fder and not use_fder2) then
    u = Functions.state1({f, u1}, time);
  else
    u = f;
  end if;
```

We use the old adaptors in FMI trainings in order to explain the coupling to acausal signals in Modelica.
I think the new ones are to complicated for the "simple" task.

Note: Working graphically, the new icons look nice and you won't recognize a difference.
But, if you look at the diagram of the adaptor itself, it's already harder to understand.

My personal proposal:
Revert to the old version.
Keep the new version with a different name, e.g. TorqueToAngleAdaptorGeneralized

tobolar, 2018-05-03:
In that case
1. the classes should be renamed - your suggestion is fine to me,
2. the different structure/advantages/disadvantages should be documented in the new adaptors,
3. an example may be added; or a notice should be added into Modelica.Mechanics.Rotational.Examples.GenerationOfFMUs (using new adaptors) that another adaptors may be used as well and
4. icons should be modified to distinguish between adaptors used.

tobolar, 2018-05-07:
I would prepare a PR.

tobolar, 2018-05-07:
I would rename the new classes for v3.2.3. Nontheless this should be considered as a temporary solution since there are meanwhile fare more adaptors extending from 
`Modelica.Blocks.Interfaces.PartialFMUadaptors.FlowToPotentialAdaptor` (or PotentialToFlowAdaptor):
- Modelica.Mechanics.Translational.Components.ForceToPositionAdaptor, PositionToForceAdaptor
- Modelica.Electrical.Analog.Basic.CurrentToVoltageAdaptor, VoltageToCurrentAdaptor
- Modelica.Thermal.HeatTransfer.Components.HeatflowToTemperatureAdaptor, TemperatureToHeatflowAdaptor

Since there should be granted naming constistency within MSL, the adaptors in Modelica.Mechanics.Rotational.Components should be renamed eventually in the next MAJOR version.
For compatibility reasons, the names of the "old" adaptors should stay untouched till then.

For the next MAJOR, we rather need suggestion for names of the "old" adaptors. I would open a new issue once this one is solved.

Yet another issue is **the icon of the partial adaptors** `FlowToPotentialAdaptor` and `PotentialToFlowAdaptor`. I suppose the icons currently base on the icon of "old" adaptors from Modelica.Mechanics.Rotational.Components - being horizontally gray shaded. This is reasonable for Rotational adaptors having a link to the rotational inertia component. But this doesn't apply for another domains. 
So what could be more reasonable? Simply an empty (rounded?) rectangle or something else?

beutlich, 2018-05-08:
> For the next MAJOR, we rather need suggestion for names of the "old" adaptors. I would open a new issue once this one is solved.

There already is Modelica.Magnetic.FundamentalWave.BasicMachines.Components.SymmetricMultiPhaseCageWinding_obsolete, which might work as an idea.

> But this doesn't apply for another domains. So what could be more reasonable? Simply an empty (rounded?) rectangle or something else?

Confirmed, CurrentToVoltageAdaptor (and others) should not be horizontally shaded. I also suggest to have an empty (rounded) rectangle here to increase the visibility of the text on top.

tobolar, 2018-05-14:
Icons: when having white-filled rounded rectangle in `FlowToPotentialAdaptor` and `PotentialToFlowAdaptor`, the adaptors extended could have rounded non-filled rectangle of particular color and maybe double thickness to distinguish between them. The result would be like in this picture (adaptors for Rotational, Thermal and Electrical are all on the right).

[image]

AHaumer, 2018-05-18:
Well, I suggest to keep both the old and the new ones. 
Just make a proposal for renaming and changing the icons.

AHaumer, 2018-05-21:
Since Leo likes the "old" variant, I'll re-introduce them in parallel to the "new" ones.

AHaumer, 2018-05-21:
Have to revert the example to the "old" version, too, and move the test of the "new" ones to ModelicaTest.

tobolar, 2018-05-28:
I don't believe f3a6bdd769d5f36ce85442b654369c8dd3cb01e0 really closes this issue because of
1. Icons: that commit introduces different icons for `FlowToPotentialAdaptor`and `PotentialToFlowAdaptor` (from Modelica.Blocks.Interfaces.PartialFMUadaptors) - being of fillPattern=FillPattern.VerticalCylinder and fillPattern=FillPattern.Sphere, respectively. This may be reasonable but my suggestion has been to have rather less dominant icons for those generic adaptors enabling to highlight domain-specific definitions of them. Thus, empty icons could be a possible way. 
In the figure above, the suggested domain-specific adaptors are clearly recognizable by a domain-specific color. So we can either have
    1. Icons reflecting domain-specific adaptors (as suggested in figure above),
    2. Icons reflecting flow to potential direction and vice versa in a general manner (as in f3a6bdd769d5f36ce85442b654369c8dd3cb01e0) or
    3. both.<br>
I would prefer the first or the third option.

2. Name of the new adaptors in `Modelica.Mechanics.Rotational.Components`. To rename `TorqueToAngleAdaptorNew` to `GeneralTorqueToAngleAdaptor` could be fine but we shoud keep in mind that this is just a temporary name since all adaptors extending the generic adaptors should use the same naming pattern. Thus, also Modelica.Mechanics.Translational.Components.ForceToPositionAdaptor, etc., should be renamed accordingly (i.e. to GeneralForceToPositionAdaptor).
Accepting GeneralTorqueToAngleAdaptor as a temporary name, the following conversion have to be done in the next major version:
   - Modelica.Mechanics.Rotational.Components.GeneralTorqueToAngleAdaptor -> TorqueToAngleAdaptor
   - Modelica.Mechanics.Rotational.Components.TorqueToAngleAdaptor -> TorqueToAngleAdaptorXXX with XXX being some text specifiing the uniqueness of the adaptor, e.g. "Flat" according to @GallLeo.

commit f3a6bdd7 (2018-05-20): resolves #2430
commit e477bcb9 (2018-05-21): resolves #2430
commit 9f5d7e88 (2018-05-21): resolves #2430
PR #2562 (2018-05-29): #2430 revised adaptor icons and nameing
Following tobolar's suggestions in:

1. Icons:
- removed shading in icons for improved clearity in partial adaptor
- added colored rounded rectangle in domain specific adaptors
- unified sizing and location of real input icons 

2. Naming:
- renamed XToYAdaptor to GeneralXToYAdaptor

close #2430
closed 2018-05-29 by AHaumer