modelica/ModelicaStandardLibrary #2432: Use ground plane visualizer directly in the World

tobolar, 2018-01-18:
Since `Modelica.Mechanics.MultiBody.Visualizers.Ground` has strong relation to `Modelica.Mechanics.MultiBody.World`, it should be a part of it.

beutlich, 2018-03-05:
Reset milestone since 66345025f76bdfbb06ac7cd33e53707a25402718 went to master branch.

HansOlsson, 2018-06-18:
This change causes regressions in a commercial library (FlightDynamics); and in theory there could be other similar cases.

Basically if a library made their own world-object (by copying World from MSL and adding parts) and used it as "inner MyWorld world;" the "outer World world;" in MSL no longer matches the inner component.

I don't see that we have stated that making your own inner-objects is not fully supported.

HansOlsson, 2018-06-18:
Oh, perhaps better to open new issue. Will do that.

commit 66345025 (2018-03-05): Closes #2432: Added ground plane visualization into the multi-body world
closed 2018-06-18 by HansOlsson