modelica/ModelicaStandardLibrary #2438: Same name R used for universal and specific gas constant

hubertus65, 2018-01-23:
The same name for the universal and specific gas constant can lead to errors, as we had in #2377. A name change of either the universal gas constant form R to R_m, or from the specific gas constant R to R_s. Both changes would touch a lot of points and could actually be disruptive for users even with good conversion scripts. Good unit checking would always detect it, good documentation could reduce errors. Therefore I'll pose this issue as discussion.

HansOlsson, 2018-01-23:
I understand that there are some cases where R is used for the specific gas constant, but I would prefer to keep R for the universal gas constant, since NIST defines it, https://physics.nist.gov/cgi-bin/cuu/Value?r and the ideal gas-law pV=nRT is fairly well-known.

I would also prefer "R_specific" over "R_s".

hubertus65, 2018-01-29:
Fine with me.

beutlich, 2019-07-15:
I started to have a closer look at the conversion effort :sweat: and noticed that we already have names like R_gas and R_s for the specific gas constant. Should we rename all such variables to R_specific or just the misleading ones where we currently have R only as ident?

beutlich, 2019-07-26:
@thorade Do you have any proposals?

hubertus65, 2019-08-21:
I's say: bite the bullet now and update all places. It could make sense to document such naming conventions at one place per sublibrary and make sure they are followed in commit audits. I have no strong opinion re R_s or R_specific.

thorade, 2019-09-06:
I agree with Hans and Hubertus to use R for universal gas constant and R_s or R_specific for the gas specific gas constant, so that 
```
R_s = R/MM
```
I would prefer R_s because it is shorter in equations, and is also in line with what I have seen in text-books, but this is not a strong preference either.

PR #3190 (2019-11-08): Rename specific gas constant R to R_s
resolves #2438
closed 2019-11-08 by beutlich