modelica/ModelicaStandardLibrary #2445: Inconsistent 'Contact' section of User's Guides of packages

tobolar, 2018-01-25:
In `Modelica.UsersGuide.Conventions.Documentation.Structure` there is defined how to format the structure of a class documentation. Consequently, the 'Contact' section of lot of the User's Guides of packages should be updated according to those definition. In particular using
```
<h4>Acknowledgements</h4>
```
instead of
```
<p><strong>Acknowledgements:</strong></p>
```
(and esp. to delete the colon!)

Moreover, there exists a design recommendation for the structure of 'Contact' class, see `Modelica.UsersGuide.Conventions.UsersGuide.Contact`. But I think that recommendation is rather odd and, as a matter of fact, it is used only in `Modelica.Magnetic.QuasiStatic.FundamentalWave.UsersGuide.Contact`.

To me, a good example on how to format 'Contact' is `Modelica.ComplexBlocks.UsersGuide.Contact`, aside from the subsection title 'Contact' which rather should be 'Library Officers`.

Maybe similar also applies to `References` but I haven't check.

beutlich, 2018-01-25:
If you are about to touch and harmonize these contact paragraphs, would you mind to also update the others sublibrary one's even though it is not you domain? Thanks a lot!

tobolar, 2018-01-25:
No problem. I hope the others agree to it.

tobolar, 2018-01-25:
I have modified this across the library and we can close this issue.
I will additionally prepare a PR with new design suggestion for `Modelica.UsersGuide.Conventions.UsersGuide.Contact`.

closed 2018-01-25 by tobolar