modelica/ModelicaStandardLibrary #2460: Shorten names of sensor models in

tobolar, 2018-02-08:
Sensor models spread across the library don't need to contain the word "sensor" in their name when collected in packages called "Sensors". Applying this rule, the sensor models' names can be shortened significantly.
I suppose the 'Sensor' in the name is hangover from former library versions to simplify naming of sensor instances. But using `defaultComponentName` consequently can make it as well.
This applies to
- `Modelica.Electrical.Analog.Sensors`
- `Modelica.Electrical.Machines.Sensors`
- `Modelica.Electrical.MultiPhase.Sensors`
- `Modelica.Electrical.QuasiStationary.SinglePhase.Sensors`
- `Modelica.Electrical.QuasiStationary.MultiPhase.Sensors`
- `Modelica.Magnetic.FluxTubes.Sensors`
- `Modelica.Magnetic.FundamentalWave.Sensors`
- `Modelica.Mechanics.MultiBody.Sensors`
- `Modelica.Mechanics.Rotational.Sensors`
- `Modelica.Mechanics.Translational.Sensors`
- `Modelica.Thermal.FluidHeatFlow.Sensors`
- `Modelica.Thermal.HeatTransfer.Sensors`

In contrast, the rule is already applied to `Modelica.Fluid.Sensors`. Despite no `defaultComponentName` is applied.

HansOlsson, 2018-02-08:
Hmm.... I agree with this change and removing the need for `defaultComponentName` for the classes. 
The shorter name removes redundant information when looking for the class.
I feel that `defaultComponentName`  is easily over-used and is non-intuitive for users, since we build up the expectation that the class-name is used as component-name, and then suddenly some other name is used.

tobolar, 2018-02-08:
I don't know how much this also concern to sensors. The motivation is simply to not repeat the name of the package in the classes contained. But in my opinion it is helpful to reflect "sensor" in the instance name.

I guess a broad agreement within MA is needed to solve this issue.

christiankral, 2019-08-26:
My opinion is that it does not make sense to have the word `Sensor` twice in the full qualified class name, e.g. `Modelica.Electrical.Polyphase.Sensors.VoltageSensor`. It yet makes sense to have `sensor` in the instance name. 

### Example

[image]

We rename `Sensors.VoltageSensor` to `Sensors.Voltage` and change the `defaultComponentName` to `voltageSensor`. In this case a user used to MSL 3.X may even not notice that the sensor class names changed, if instance names are still same as before.

christiankral, 2019-08-26:
@tobolar Do you think a formal broad agreement within MAP-LIB is needed here? Shall we make a poll instead using :+1: and :-1: on this comment?

tobolar, 2019-09-02:
> Do you think a formal broad agreement within MAP-LIB is needed here?

This could be reasonable if we want to establish this as a common rule for developers. Imagine how this could simplify e.g. the following path: `Modelica.Electrical.Machines.BasicMachines.SynchronousMachines.SM_PermanentMagnet` to `Modelica.Electrical.Machines.Basic.Synchronous.SM_PermanentMagnet`. ;-)

tobolar, 2020-01-28:
Due to missing further discussion we'll not be able to fix this by feature freeze.