modelica/ModelicaStandardLibrary #2465: Modelica.Mechanics.MultiBody.Joints.Prismatic - s_offset - constant vs parameter

natehorn, 2018-02-12:
Is there any reason that s_offset is a constant rather than a parameter?  Thanks!

beutlich, 2018-02-12:
This lines
https://github.com/modelica/Modelica/blob/1de3f959ddaff0c106a9e0fcb56af61ff54b25ac/Modelica/Mechanics/MultiBody/Joints.mo#L22-L24

FrenkelJ, 2018-02-12:
By the way the Evaluate Annotation is only specified for a component declared with the prefix parameter. See Spec. 18.3 Evaluate.

beutlich, 2018-02-13:
It looks like a bug introduced by c025e058b032a9542ceba9ca0335a6a59ed6da00. Note that `s_offset` is a proper parameter (without Evaluate annotation) in the other aggregate joints with prismatic degree of freedom.

MartinOtter, 2018-02-13:
No, this is not a bug, but a feature. The reason is that I wanted to get rid of offset, to make joints simpler. For backwards compatibility reasons, "offset" was not removed but kept. Via "constant" it was, however, removed from the parameter menu. Note, a constant "phi_offset" is also present in the Revolute joint (for the same reasons).

beutlich, 2018-02-13:
- Still it would be wise to remove the Evaluate annotation for MSL 3.2.3.
- Please file a new ticket with milestone MSL_next-MINOR-version to remove the superfluous `s_offset` constant then.

tobolar, 2018-02-19:
Another option is to let `s_offset` be a parameter with `annotation (Dialog(enable=false))`. This enables a user to still see there is such a parameter with some value which influences the model behavior.

tobolar, 2018-02-19:
@beutlich 
> ... new ticket with milestone MSL_next-MINOR-version ...

Your rather mean MSL_next-MAYOR-version?

closed 2018-02-21 by tobolar

beutlich, 2018-02-21:
> Your rather mean MSL_next-MAYOR-version?

Yes, edited MSL_next-MAJOR-version above. Thanks.

commit 56bda294 (2018-02-21): Bug fix #2465 : removed the annotation 'Evaluate' in constant s_offset of Modelica.Mechanics.MultiBody.Joints.Prismatic