modelica/ModelicaStandardLibrary #2466: Question on behavior of OneWayClutch

UlfNordstrom, 2018-02-13:
Hi,
One of our customers is asking about the behavior of the Modelica.Mechanics.Rotational.Components.OneWayClutch and I hope you could help clarify.

The remark is that OneWayClutch 
"should transmit rotational force only in a fixed direction. In present implementation in MSL, it transmits rotational force as follows"
<pre>
                |   true   |   false
Torque positive | transmit |  transmit
Torque negative | transmit |  No transmit
</pre>

"OneWayClutch should not convey rotation in both directions. It conveys rotation in one direction.
Also, since it is Clutch, it should not transmit torque when "false". Therefore OneWayClutch should work as follows"
<pre>
                |   true   |   false
Torque positive | transmit |  No transmit
Torque negative | transmit |  No transmit
</pre>

Could you please comment on this?

Best regards,
Ulf

AHaumer, 2018-02-16:
Maybe Martin O. should comment on this, I think he designed the component. 
IMHO there's an ambiguity in the decoumentation regarding series or parallel connection of the clutch and the freewheel. Citing the documentation:
"When the relative angular velocity becomes zero, the elements connected by the friction element become stuck. ... The elements begin to slide when the friction torque exceeds a threshold value."
Keep this in mind, and explore the behavior with a simple test model (enclosed). 
Change the frequency of the torque source, and observe oneWayCluth.{w_rel, a_rel, tau}.
For me it seems like a parallel connection of clutch and freewheel. 
[TestOneWayClutch.zip](https://github.com/modelica/Modelica/files/1732036/TestOneWayClutch.zip)

tobolar, 2018-02-19:
Regarding
>an ambiguity in the decoumentation regarding series or parallel connection of the clutch

see #2467.

tobolar, 2018-02-21:
@AHaumer I have adapted your model for lower torque frequency in the hope that the functionality of the clutch would be even better understandable.
[TestOneWayClutch.zip](https://github.com/modelica/Modelica/files/1743580/TestOneWayClutch.zip)

I would add this example to `Rotational.Examples` if no objections arise.

tobolar, 2018-02-21:
@UlfNordstrom To respond to your question: The one-way-clutch model represents a clutch and a free wheel connected in parallel way. When the clutch is disengaged (i.e. signal f_normalized = 0) it behaves like the free-wheel. Once the clutch is engaged it transmits a torque independent of the torque/rotation direction.

The second scheme in your question reflects behaviour of comon clutch.

UlfNordstrom, 2018-02-22:
Hi,
Thank you for you answer. I have relayed to our customer.
Best regards,
Ulf

UlfNordstrom, 2018-02-26:
Hi,

Our customer have some follow up questions.

They have understood the present MSL One Way Clutch behavior, and we found a mistake when the text was translated to English:

 "Expected Result". The following is the corrected expected result. 
<pre>
                |   true      |   false
Torque positive | transmit    |  No transmit
Torque negative | No transmit |  No transmit
</pre>

But according to the customer, this is not the normal behaviour of a one way clutch, as in this Catalog http://www.ntnglobal.com/en/products/catalog/pdf/2900E.pdf .

Is it possible to realize the "expected result" with the current models of MSL?

Best regards,
Ulf

tobolar, 2018-02-26:
@UlfNordstrom Maybe there is a misunderstanding in the customer's table since in my opinion the one-way-clutch behavior is given by the relative speed between output and input shafts, which always is >= 0. To my understanding, this principle is also described in the Catalog you have mentioned, in Section 1-2 Operating mechanism on page 4.

The example provided by @AHaumer also shows behaviour similar to that shown on page 19 of the Catalogue.
Could your customer provide a simple example as well?

tobolar, 2018-03-08:
Dear @UlfNordstrom, any progress on this issue?

beutlich, 2018-03-09:
@tobolar Both models added by 441f046d03032e9ccc78c919df2a2276d91ee04d give two warnings about unspecified initial systems.

```
Warning In Modelica.Mechanics.Rotational.Examples.OneWayClutch(Disengaged)
For the variable "_pre_oneWayClutch.stuck" is no assignment in the initial system. The startvalue "oneWayClutch.stuck.start" is used. Set fixed=true or add an assignment in an initial section
Message: InitialSystem.MissingInitialAssignment

Warning In Modelica.Mechanics.Rotational.Examples.OneWayClutch(Disengaged)
For the variable "_pre_oneWayClutch.startForward" is no assignment in the initial system. The startvalue "oneWayClutch.startForward.start" is used. Set fixed=true or add an assignment in an initial section
Message: InitialSystem.MissingInitialAssignment
```

tobolar, 2018-03-12:
@beutlich Sorry for this bothering mistake. I'm going to fix it.

tobolar, 2018-03-29:
Closed since no new circumstances occured in the last weeks.

closed 2018-03-29 by tobolar