modelica/ModelicaStandardLibrary #2470: Wrong link order of Libraries used by Modelica.Blocks.Tables

rfranke, 2018-02-23:
Modelica.Blocks.Tables.CombiTable{1D,1Ds,2D} contain external function calls and
```
annotation (Library={"ModelicaStandardTables", "ModelicaMatIO", "zlib"});
```
This is the wrong link order, because ModelicaMatIO uses functions of ModelicaStandardTables (Mat_Open, Mat_VarReadInfo, ...).

**The annotation should be**:
```
annotation (Library={"ModelicaMatIO", "ModelicaStandardTables", "zlib"});
```
The problem can be reproduced with OpenModelica's FMI export under Windows using a model like
```
model TableTest
  extends Modelica.Blocks.Tables.CombiTable2D(
      table = [0.0,   1.0,   2.0,   3.0;
               1.0,   1.0,   3.0,   5.0;
               2.0,   2.0,   4.0,   6.0]);
  annotation(uses(Modelica(version="3.2.2")));
end TableTest;
```
OpenModelica contains MinGW. FMI export shall produce a dll that contains all external dependencies.  Using the bundled g++ version 5.3.0, the linker call is as follows,
```
g++ -shared -o TableTest.dll OMCppTableTestCalcHelperMain.o ... -Wl,--no-undefined ...\
    -lModelicaStandardTables -lModelicaIO -lModelicaMatIO -lzlib ... -static
```

This produces the error:
```
C:/OpenModelica1.13.0-dev-32bit//lib//omc\libModelicaIO.a(ModelicaIO.o): In function `readMatIO':
C:\dev\OpenModelica32bit\OMCompiler\3rdParty\ModelicaExternalC\BuildProjects\autotools/../../C-Sources/ModelicaIO.c:446: undefined reference to `Mat_Open'
...
collect2.exe: error: ld returned 1 exit status
TableTest_FMU.makefile:93: recipe for target 'TableTest.fmu' failed
make: *** [TableTest.fmu] Error 1
```

The problem is worked around in the current nightly build of OpenModelica by linking ModelicaStandardTables a second time after the model defined external libs, see:

https://github.com/OpenModelica/OMCompiler/commit/cacc8bf978c0be0acf4e3197159ca1dd328f494d

It should be solved by changing the order of libraries in Modelica.Blocks.Tables.*.

beutlich, 2018-02-23:
> This is the wrong link order, because ModelicaMatIO uses functions of ModelicaStandardTables (Mat_Open, Mat_VarReadInfo, ...).

Hm, Mat_Open and Mat_VarReadInfo are declared in module ModelicaMatIO.c and utilized by module ModelicaStandardTables.c. Thus, the other way round.

Can you check with the OMC linking again with current master, especially after my 11.5 months old commit 7ac5e517d594601983f42fe008f43970188b4bb6 for #2192?

beutlich, 2018-02-23:
I also wonder why OMC reports this linking error right now, after I added it more than 2 years ago (see f9a30ca2e0a3a4be6e2261185c006ad01e456480 for #1856).

sjoelund, 2018-02-23:
I suppose because we mostly test OpenModelica on Linux which uses dynamic linking, and there all so-files are already linked against their respective symbols.

rfranke, 2018-02-23:
> Hm, Mat_Open and Mat_VarReadInfo are declared in module ModelicaMatIO.c and utilized by module ModelicaStandardTables.c. Thus, the other way round.

If the functions are defined in ModelicaMatIO and utilized by ModelicaStandardTables, then the annotation should indeed be changed to:
```
annotation (Library={"ModelicaMatIO", "ModelicaStandardTables", "zlib"});
```

sjoelund, 2018-02-23:
If they are defined in ModelicaMatIO but used by Tables, Tables should be before on the list, just like zlib is last because the functions are defined in zlib and used by ModelicaMatIO.

rfranke, 2018-02-23:
Yes, you are right, sorry. 

This would mean that OpenModelica messes up C source names and libarary names, because libModelicaMatIO does not have the functions (link order `-lModelicaStandardTables -lModelicaMatIO`):
```
C:/OpenModelica1.13.0-dev-32bit//lib//omc/cpp\libModelicaMatIO.a(ModelicaIO.c.obj):ModelicaIO.c:(.text+0xf0): undefined reference to `Mat_Open'
C:/OpenModelica1.13.0-dev-32bit//lib//omc/cpp\libModelicaMatIO.a(ModelicaIO.c.obj):ModelicaIO.c:(.text+0x120): undefined reference to `Mat_VarReadInfo'
```
The linker error disappears with reversed link order  `-lModelicaMatIO -lModelicaStandardTables`.

sjoelund, 2018-02-23:
I would suggest trying to compile with the actual libraries generated by the ModelicaExternalC project instead of [the custom CMakeLists.txt](https://github.com/OpenModelica/OMCompiler/blob/master/SimulationRuntime/cpp/Core/ModelicaExternalC/CMakeLists.txt) used by the Cppruntime. ModelicaIO doesn't seem to have its own library there which is sure to cause some problems. The ModelicaMatIO.a only contains ModelicaIO.c as well... Just crazy.

beutlich, 2018-02-23:
OK, closing as invalid then.

closed 2018-02-23 by beutlich

rfranke, 2018-02-23:
Sorry for having spammed this tracker with OpenModelica issues. I couldn't imagine that source names and library names differ. The following commit should clean it up: https://github.com/OpenModelica/OMCompiler/commit/90621756a8e8f3978a17347146a6fc3b044f3e71