modelica/ModelicaStandardLibrary #2472: New block LimitedPI vs. existing LimPID

beutlich, 2018-02-25:
2c8f8a2cf1d65065bf61f31364829bcf821175dd added a new block Modelica.Electrical.Machines.Utilities.LimitedPI but rather should utilize (or even update) the existing block Modelica.Blocks.Continuous.LimPID for the sake of reusable component models (instead of fragmented utilities models).

AHaumer, 2018-02-25:
Please take a closer look before opening an issue.
There are rather big differences compared with the existing block, as these features are needed in controls of electric drives:
* The new block has variable limits.
* The new block has an optional feedforward-input.

dietmarw, 2018-02-25:
And those features can not be added to the already existing block. And if not why not.

AHaumer, 2018-02-25:
I think the LimPID block should be kept as it is without overloading it, since lots of users use it as it is. 
Additionally I don't need the D-part which makes initialization easier. 
Therefore I preferred to cerate the LimPI (with variable limits and feedforward). 
According to your argument we should mark PI and PID as obsolete since they can be replaced by LimPID.

beutlich, 2018-02-25:
Two more comments

- There also is Modelica_LinearSystems2.Controller.LimPID, another PID controller with limited output. I always do wonder why we need same basic component models in all places.

- There is Modelica.Blocks.Nonlinear.Limiter (for parameter limits) and Modelica.Blocks.Nonlinear.VariableLimiter (for variable limits). I see no reason not to add a new block VariableLimPI(D) to Blocks.

AHaumer, 2018-02-25:
Modelica_LinearSystems2 is not (yet?) part of the MSL. 
Modelica_LinearSystems2 shall ease swicthing beetween continuous and duscrete controllers.

Yes there are two limiters in Blocks. The VariableLimiter is used in the LimitedPI.
But the LimitedPI is more than a limiter ... don't compare blocks with different goal. 
You could also argue to remove components that are composed from simpler components,

beutlich, 2018-02-26:
3fd3d6a736386e9c59a13c61559b38293d62d929 (silently) moved Modelica.Electrical.Machines.Utilities.LimitedPI to Modelica.Electrical.Machines.Examples.ControlledDrives.Components.Limited which basically resolves the issue. Still there is #2475 on the name.

closed 2018-02-26 by beutlich