modelica/ModelicaStandardLibrary #2474: Spherical, etc. render poorly on non-white backgrounds

tobolar, 2018-02-26:
Basically this is a copy of #2031 but this time for `Modelica.Mechanics.MultiBody` components, as e.g. for
- Modelica.Mechanics.MultiBody.Joints.Spherical,
- Modelica.Mechanics.MultiBody.Joints.SphericalSpherical,
- Modelica.Mechanics.MultiBody.Joints.UniversalSpherical or
- components from Modelica.Mechanics.MultiBody.Joints.Assemblies.

beutlich, 2018-02-28:
Sorry Jakub, I needed to revert your commit 79ac51b48fce08e323d1da95186c2f8fd0d0a4ff since it introduced features of Modelica 3.4, which must not be used in master branch targeting Modelica 3.2 for now.

Not sure if this means the rendering cannot be improved with Modelica 3.2 language then, but to me it looks this way. Anyway, you are free to open an new pull request targeting NEXT_MAJOR milestone but I am afraid it will not be merged too soon then.

tobolar, 2018-03-01:
@beutlich Ok, sorry for this annoyance. 
In Modelica 3.2 specification both the start and end angle for ellipse primitive are defined. So this should not be the problem. In contrast, _closure_ is first defined in Modelica 3.4.

I will prepare a pull request.

beutlich, 2018-03-01:
True. But it all depends if you like the default closure using chords then.

tobolar, 2018-03-01:
Not only if I like it. :-)

tobolar, 2018-03-01:
The incompatibility to Modelica 3.2 also applies to my  ba030cb7149226a39082d405a0aec2f341c85971. I'm going to commit a fix of `Torque` component.

tobolar, 2019-09-03:
@beutlich
Can I prepare a PR for this? In MSL, it is still stated: It is fully compliant to Modelica Specification Version 3.2 Revision 2 ...

beutlich, 2019-09-03:
> Can I prepare a PR for this?

Yes, pleasego ahead.

> In MSL, it is still stated: It is fully compliant to Modelica Specification Version 3.2 Revision 2 ...

Right, this was not yet updated.

beutlich, 2019-11-04:
@tobolar What is your time schedule regarding MSL v4.0.0?

PR #3634 (2021-01-20): Utilize ellipses' closure
Closure introduced in Modelica Specification 3.4
closes #2474
closed 2021-01-20 by tobolar